arduino/Arduino #2779: Arduino 1.6.0-windows exe. error = `error writing to file lto 1.exe`

Maurgo, 2015-03-16:
Arduino 1.6.0-windows exe.  error = `error writing to file lto 1.exe`.

While trying to install the above on my windows 8.1 machine I get this error. It has installed on Windows 7 fine. Can anyone help please?   thanks

ffissore, 2015-03-16:
Looks an issue with your windows installation. Have you tried 1.6.1 ?

Maurgo, 2015-03-16:
Hello ffissre, yes I have tried 1.6.1 - the same.  I think too it may be windows installation, I have done a check and is up to date so I do not know which file to check?

ffissore, 2015-03-16:
Could it be a permission error? A full disk? Have you tried using the zip version instead of the installer (exe) ?

Maurgo, 2015-03-16:
I dont know to be honest!   No I have not tried the zip edition, think I will give that a go

closed 2015-05-22 by agdl