arduino/Arduino #2782: Tutorial/ReadASCIIString

logus2000, 2015-03-17:
Hi, I found that this tutorial didn´t work, I trace the problem to a simpre bracelet out of place so the program should be like this

*
  Reading a serial ASCII-encoded string.
 
 This sketch demonstrates the Serial parseInt() function.
 It looks for an ASCII string of comma-separated values.
 It parses them into ints, and uses those to fade an RGB LED.
 
 Circuit: Common-anode RGB LED wired like so:
 * Red cathode: digital pin 3
 * Green cathode: digital pin 5
 * blue cathode: digital pin 6
 * anode: +5V
 
 created 13 Apr 2012
 by Tom Igoe
 
 This example code is in the public domain.
 */

// pins for the LEDs:
const int redPin = 3;
const int greenPin = 5;
const int bluePin = 6;

void setup() {
  // initialize serial:
  Serial.begin(9600);
  // make the pins outputs:
  pinMode(redPin, OUTPUT); 
  pinMode(greenPin, OUTPUT); 
  pinMode(bluePin, OUTPUT); 

} // this was at the end of the program, I replace it here

void loop() {
  // if there's any serial available, read it:
  while (Serial.available() > 0) {

    // look for the next valid integer in the incoming serial stream:
    int red = Serial.parseInt(); 
    // do it again:
    int green = Serial.parseInt(); 
    // do it again:
    int blue = Serial.parseInt(); 

    // look for the newline. That's the end of your
    // sentence:
    if (Serial.read() == '\n') {
      // constrain the values to 0 - 255 and invert
      // if you're using a common-cathode LED, just use "constrain(color, 0, 255);"
      red = 255 - constrain(red, 0, 255);
      green = 255 - constrain(green, 0, 255);
      blue = 255 - constrain(blue, 0, 255);
      
}
      // fade the red, green, and blue legs of the LED: 
      analogWrite(redPin, red);
      analogWrite(greenPin, green);
      analogWrite(bluePin, blue);

      // print the three numbers in one string as hexadecimal:
      Serial.print(red, HEX);
      Serial.print(green, HEX);
      Serial.println(blue, HEX);
    
  }
}

agdl, 2015-04-03:
@logus2000 why doesn't work in your opinion? Whenyou send 3 int values separated by a comma the color change

agdl, 2015-04-03:
@logus2000 the example included in the IDE works. Why do you state that it doesn't work? The parenthesis you refer to is in the right place!

logus2000, 2015-04-04:
maybe you need to make it fisical instead of running the program in your mind, I run the program by copying and paste out of this site, not from the IDE
http://arduino.cc/en/Tutorial/ReadASCIIString
if it runs with you send me a video please.

agdl, 2015-04-09:
First of all I made it and it works. In second place please keep calm because I'm just trying to answer to your request.

The sketch is included in the IDE and so I ran it from there since you stated no where that you used that one form the site (that by the way makes reference to the Git link -> so to the IDE)

The sketch works. The parenthesis that you commented:

} // this was at the end of the program, I replace it here

is in the right place and the sketch is the same apart from another parenthesis that is moved in another place:

 while (Serial.available() > 0) {

    // look for the next valid integer in the incoming serial stream:
    int red = Serial.parseInt(); 
    // do it again:
    int green = Serial.parseInt(); 
    // do it again:
    int blue = Serial.parseInt(); 

    // look for the newline. That's the end of your
    // sentence:
    if (Serial.read() == '\n') {
      // constrain the values to 0 - 255 and invert
      // if you're using a common-cathode LED, just use "constrain(color, 0, 255);"
      red = 255 - constrain(red, 0, 255);
      green = 255 - constrain(green, 0, 255);
      blue = 255 - constrain(blue, 0, 255);

}

instead of:

 while (Serial.available() > 0) {

    // look for the next valid integer in the incoming serial stream:
    int red = Serial.parseInt();
    // do it again:
    int green = Serial.parseInt();
    // do it again:
    int blue = Serial.parseInt();

    // look for the newline. That's the end of your
    // sentence:
    if (Serial.read() == '\n') {
      // constrain the values to 0 - 255 and invert
      // if you're using a common-cathode LED, just use "constrain(color, 0, 255);"
      red = 255 - constrain(red, 0, 255);
      green = 255 - constrain(green, 0, 255);
      blue = 255 - constrain(blue, 0, 255);

      // fade the red, green, and blue legs of the LED:
      analogWrite(redPin, red);
      analogWrite(greenPin, green);
      analogWrite(bluePin, blue);

      // print the three numbers in one string as hexadecimal:
      Serial.print(red, HEX);
      Serial.print(green, HEX);
      Serial.println(blue, HEX);
    }
  }

so I can't understand where is your problem.

agdl, 2015-04-09:
Have you the new line selected ad the end of the output? You can select it beside where you can choose the serial speed.
Please let me know.

agdl, 2015-04-16:
Reopen this if needed

closed 2015-04-16 by agdl