arduino/Arduino #2784: Burning uploader need a short delay

ffissore, 2015-03-17:
As reported here (italian) http://forum.arduino.cc/index.php?topic=153592.msg2140757#msg2140757

matthijskooijman, 2015-03-17:
This seems to be programmer-specific. The report is for an AVRISP MKII, and I think the post says it doesn't happen for the dragon. I have a JTAGICE3 and STK500, which both burn bootloaders just fine with the current code.

darkwingz24, 2015-04-24:
I have an AVRISP MKII and have the issue reported. Error says " avrdude: usbdev_open(): did not find any USB device "usb" ". I have the libusb drivers installed and it works under old versions of Arduino but since I upgraded to 1.6.x burning bootloaders from AVRISP MKII has been broken. I can see the exposed LED go from Green to orange and the internal green led blink after I hit "burn bootloader" but then after half a second it reports the error above.  I keep clicking "burn bootloader" and sometimes it works (1 out of 20 maybe). Mostly it doesn't work. Happens on Arduino 1.6.x on Win 7x64 and Win 8.1x64 machines.

per1234, 2015-09-08:
There is further discussion of this issue in #2986

per1234, 2016-09-24:
@cmaglie this is a duplicate of #2986.
Anyone who has previously experienced this issue please test with Arduino AVR Boards 1.6.14/Arduino IDE 1.6.12 to see if the issue still occurs with avrdude 6.3 and report your findings at #2986. It fixed the issue for me but not for bperrybap.

closed 2017-07-01 by per1234