arduino/Arduino #2787: Tools-Menu is not appeared.

sunaokumura, 2015-03-18:
Arduino IDE Ver.1.6.1 on Windows7 64bit and I use it on Java 8.
Tools-Menu is not appeared.
Text Area is small.

I tried following:
http://arduino.cc/download.php?f=/javaide/pull_requests/arduino-PR-2783-BUILD-224-windows.zip
but This Arduino IDE has same issue as Ver.1.6.1.

[image]

sunaokumura, 2015-03-18:
and Arduino IDE Ver.1.0.6 has no problem as below.

[image]

ffissore, 2015-03-18:
Is there any customization you made to windows that could help reproduce your issue? I too have a windows7 64bit and the build from #2783 shows fine :(

sunaokumura, 2015-03-18:
There is no customization.
I tried #2783 again.
Thank you!

ffissore, 2015-03-19:
Fixed/Improved by #2783

closed 2015-03-19 by ffissore