arduino/Arduino #2789: FTDI FT800 library not working after an update to 1.6.1

BoxusES, 2015-03-18:
The FTDI FT800 library "FTDI_V1.2.0_09072014(FT800).zip"  is not working well after an update of Arduino to 1.6.0 /1.6.1 (many program errors).
My program compile without any errors with the version 1.0.4.
How to correct the compile errors issue? (may be a preference setting or other)
Some others libraries compile without error (rc-switch library by exemple).

Thank You for your assistance.
Regards.

facchinm, 2015-04-14:
Hi @BoxusES ,
the library uses some deprecated types (like `prog_char`) which need to be changed to `const char` and so on.
Here is a reference https://github.com/arduino/Arduino/wiki/1.6-Frequently-Asked-Questions#errors-related-to-avr-progmem-changes that shows how to fix them

Given the important size of the library the smartest way to fix all the problems is executing the following commands in the library's directory (Linux and MACOSX only)
``` bash
find . -type f -exec sed -i "s/uchar/char/g" {} \;
find . -type f -exec sed -i "s/prog_/const /g" {} \;
find . -type f -exec sed -i "s/PROGMEM char/PROGMEM const char/g" {} \;
```
Some errors may still be present but they should be easily understandable (like duplicate `const` )

BoxusES, 2015-04-14:
Hi Martino,

Thank You so much for your response.

 

I use "#define prog_uchar  const unsigned char

 and "#define prog_char const char

  at the beginning of the prog and it is work well.

 

But, I have a last compiler error with the following line :

PROGMEM char *info[] = {  "FT800 ImageViewer Application",

                          "APP to demonstrate interactive Jpeg decode,",

                          "using Blend function, Bitmap flip,", 

                          "& Jpeg decode"

                       };

 

With " PROGMEM const char *info[] = …" was not working.

(Arduino v1.6.3)

I am not a guru with arduino, I appreciate your help!

 

Thank You for your assistance.

 

Regards.

 

BoxusES

 

Aucun virus trouvé dans ce message.
Analyse effectuée par AVG - www.avg.fr
Version: 2015.0.5863 / Base de données virale: 4328/9527 - Date: 13/04/2015

facchinm, 2015-04-14:
Glad I could help :smile: 
To solve the last issue, try changing the line this way
```C
const char * const info[] PROGMEM =
```
it's somehow disturbing but should work! 
Regards

M

closed 2015-04-14 by facchinm