arduino/Arduino #2791: IRremote.h error in 1.6.0 & 1.6.1

jmcos, 2015-03-19:
I get the following compilation error when referencing IRremote.h:

Arduino: 1.6.1 (Windows 7), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\libraries\RobotIRremote\src\IRremoteTools.cpp:5:16: error: 'TKD2' was not declared in this scope

 int RECV_PIN = TKD2; // the pin the IR receiver is connected to

                ^
Error compiling.

PaulStoffregen, 2015-03-19:
This was supposed to be fixed in 1.6.1.

Can you please post a copy of the sketch you're compiling?

Are you absolutely certain you installed the IRremote library in a location Arduino can access?

jmcos, 2015-03-19:
int receiverpin = 11;#include <IRremote.h>
IRrecv irrecv(receiverpin);
decode_results results;void setup() {
Serial.begin(9600);
irrecv.enableIRIn();
}void loop() {
if (irrecv.decode(&results))
{
 Serial.print(results.value, HEX);
 Serial.print(" ");
 irrecv.resume();
}

&mdash;
 
Man, 63, Avoids Wrinkles
63 Yr Old Man Shares Simple DIY Skin Tightening Method He Uses At Home
http://thirdpartyoffers.juno.com/TGL3131/550ad25f980e4525f34d7st04vuc

ffissore, 2015-03-19:
It compiles fine with 1.6.1 and the latest IRRemote library from https://github.com/shirriff/Arduino-IRremote

ffissore, 2015-03-19:
Actually, a final closing curly brace is missing, but, once added, it compiles fine

jmcos, 2015-03-19:
Federico,   I downloaded the latest IRRemote library from the website, unzipped it and placed the header file in the correct directory.  The file is dated 3/10/2015 1:54 PM.  I recompiled and got the same error.  I opened IRremote.h and couldn't find TKD2. James

 
What's your flood risk?
Find flood maps, interactive tools, FAQs, and agents in your area.
http://thirdpartyoffers.juno.com/TGL3131/550b211d8fdfe211d2735st02duc

PaulStoffregen, 2015-03-19:
What *exactly* do you mean by "and placed the header file in the correct directory".

You're almost certainly installing IRremote incorrectly.  You're supposed to unzip the whole thing into Documents/Arduino/libraries.  In that folder is a small text file with instructions.

This isn't an issue with Arduino.  Version 1.6.0 had trouble.  1.6.1 is fixed.  This issue should be closed, and end-user tech support should be done via the forum.

agdl, 2015-04-09:
Like @PaulStoffregen said should be closed and the issue should be discussed on the forum in case of necessity

closed 2015-04-09 by agdl