arduino/Arduino #2796: Add 250000 baud rate in serial monitor

hallard, 2015-03-20:
Hi,

Would it be possible to add 250000 baud rate (may be 1Mb could help also) setting in serial monitor ?

250000 works whatever crystal value is (4MHz, 8MHz, 16MHz) and always with 0% error. I'm using  4Mhz low power boards and I'm stuck with 38400 because all others won't work due to slow speed CPU. 250000 work like a charm but it's not on the list
You can see explanations of my needs [here](http://hallard.me/ulpnode-bootloader-1/#upload-right)

Thanks

ffissore, 2015-05-22:
Fix available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-05-22 by ffissore

hallard, 2015-05-22:
Frederico,

250Kbps + Serial Monitor disabled during upload, that's christmas before time, I owe you a beer ;-)

ffissore, 2015-05-22:
And it's also Friday! :D I'll hope we'll have a chance to drink that beer ;)

commit 9c02f932 (2015-05-22): Serial monitor: added 230400 and 250000 baud rates. Fixes #2796 and #341