arduino/Arduino #2800: .a linkage for libraries via library.properties

NicoHood, 2015-03-20:
It would give developers a lot of more possibilities if libraries could be linked via an .a file.
So only the needed .cpp files get compiled like the core or the .ino file directory.

It would also break existing libraries. So my suggestion is just to only compile through .a if a special property was set.

You could add this perfect into such a file with an entry "linkage"
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/SoftwareSerial/library.properties

Also this link sais if i put the library into an "src/" folder it gets linked in the .ino path. That is wrong because then it would be linked via .a file. And also this option is not working anymore. Also the import function is under "sketch" not "tools".
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#source-code
it would be cool to make it work like this though.

matthijskooijman, 2015-03-21:
I like this idea of opting in to linking through a .a file through library.properties. It might even be a good idea if _some_ source files can be linked through .a files and some directly included in the link, for maximum flexibility. The reason for wanting to go through a .a file, is so you can leave out a source file from the link when it is unused, _including_ any ISRs (which are never removed by `gc-sections` once they're included, due to the `used` attribute).

> Also this link sais if i put the library into an "src/" folder it gets linked in the .ino path. This option is not working anymore. Also the import function is under "sketch" not "tools".

I'm not sure what you are saying here, but the "also" suggests they are separate issues. Please don't complicate discussion by chunking together unrelated issues, better open multiple issues instead (or, for things like what option goes where, collect opinions on the developer mailing list first).

NicoHood, 2015-04-02:
Has `* Upgraded toolchains: statically compiled, ship with Link Time Optimization plugin (disabled by default)` something to do with this issue? Any chance to see this in the future? With the new library system this would really make sense to include the .a linkage.

And I also like the idea of mathijs to link different files "normal" or via .a file.

NicoHood, 2015-08-16:
Any chance to see this soon? It would fix a lot of issues with 2 of my libraries and newer libs can be optimized better.

I'd do it myself but I dont know where to start with the whole Arduino IDE source and I also dont have enough knowledge about the linking process.

NicoHood, 2015-08-18:
I am currently trying to implement this feature myself.
After some time of research and the help of @matthijskooijman I finally got a working version that compiles the library via .a file instead. And it seems to work.

One quick question:
Why do you normally use [recursiveCompileFilesInFolder()](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/debug/Compiler.java#L1092) [to compile libraries with a /src path](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/debug/Compiler.java#L1073)?

Can't we use [compileFiles()](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/debug/Compiler.java#L661) instead and pass a `true` instead of a false? Whats the difference?

If the recursive addition is somehow needed I'd need to add a 2nd version which returns the list of files again (instead of void and adding it to a global file list). Is there any header in java, or function prototype needed or can I just place a new function inside the .java file?

What I still need to figure out is how to read in a setting from the libraries.properties file (too keep compatibility with other, older libraries).

ffissore, 2015-09-23:
Issue moved to https://github.com/arduino/arduino-builder/issues/10

closed 2015-09-23 by ffissore