arduino/Arduino #2802: Command line tool requires X to verify or upload?

jacobrosenthal, 2015-03-22:
Trying to docker the arduino command line tools. Most stuff works
```bash
[ root@a80a54ab13f1:~ ]$ /usr/local/share/arduino/arduino --get-pref pumpkins                                                             
[ root@a80a54ab13f1:~ ]$ echo $?
4
[ root@a80a54ab13f1:~ ]$ /usr/local/share/arduino/arduino --get-pref board   
uno
[ root@a80a54ab13f1:~ ]$ echo $?
0
[ root@a80a54ab13f1:~ ]$ 
```
But all verify or upload require X?
```bash
[ root@a80a54ab13f1:~ ]$ /usr/local/share/arduino/arduino --upload /root/tester/tester.ino --board arduino:avr:uno --port /dev/tty.usbmodem1411
Exception in thread "main" java.awt.HeadlessException: 
No X11 DISPLAY variable was set, but this program performed an operation which requires it.
	at sun.java2d.HeadlessGraphicsEnvironment.getDefaultScreenDevice(HeadlessGraphicsEnvironment.java:82)
	at processing.app.Base.nextEditorLocation(Base.java:476)
	at processing.app.Base.<init>(Base.java:245)
	at processing.app.Base.main(Base.java:162)
```

jacobrosenthal, 2015-03-22:
So was able to use xvfb on Synthbread in irc's advice
https://linuxmeerkat.wordpress.com/2014/10/17/running-a-gui-application-in-a-docker-container/

Seems like command line tools shouldnt need a gui though :)

closed 2015-03-22 by jacobrosenthal

ffissore, 2015-03-22:
Indeed it shouldn't. We'll fix it in one of upcoming releases

matthijskooijman, 2015-03-22:
For reference, this was already reported as #1981.