arduino/Arduino #2804: strings.length() get incorrect length in symbols of strings in two byte coding.

listores, 2015-03-22:
In arduino IDE v 1.6.1 strings.length() get incorrect length in symbols of strings in two byte coding.

For example:
[code]
#include <SoftwareSerial.h>

String StringA="SymbolsInEnglish"; //16 symbols in english
String StringB="Буквы_По-Русски_"; //16 symbols in russian (not all ^)
String StringC="English__Русский"; //16 symbols: 9 symbols in english an 7 in russian (english first)
String StringD="Русский__English"; //16 symbols: 9 symbols in english an 7 in russian (russian first)

int LengthA,LengthB,LengthC,LengthD;
char bufferA[32];

void setup() {
  LengthA=StringA.length();
  LengthB=StringB.length(); 
  LengthC=StringC.length(); 
  LengthD=StringD.length(); 
  
  Serial.begin(19200);
  
  Serial.print(StringA);
  Serial.print(":");
  Serial.println(LengthA);

  Serial.print(StringB);
  Serial.print(":");
  Serial.println(LengthB);

  Serial.print(StringC);
  Serial.print(":");
  Serial.println(LengthC);

  Serial.print(StringD);
  Serial.print(":");
  Serial.println(LengthD);
}

void loop() {
}

[/code]

Results in port monitor:

[image]

String.setCharAt() and String.CharAt() same put/get symbols in incorret places in string.
By byte number, i think...

PS: OS Windows XP SP3.

matthijskooijman, 2015-03-22:
I guess this is somewhat intentional - length returns the number of bytes in the string, none of Arduino is utf8-aware right now (and I think there would be too much overhead to make it so...).

I'm not sure if this is properly documented, though.

q2dg, 2015-03-23:
Arduino boards only uses ASCII characters. It is documented in many places along website.

ffissore, 2015-03-24:
String uses char, so one byte per each letter. You should use `wchar_t` for unicode support, although I don't know how well (or if) it works on MCUs
Closing as wontfix

closed 2015-03-24 by ffissore