arduino/Arduino #2813: ld returned 5 exit status

laisan86, 2015-03-24:
When I compile my sketch by 1.6.1, the message always showed up! No matter compile only or upload mode.
[image]

I don't know what's 'ld', so suppose you can tell me how to solve this problem. Thanks!

laisan86, 2015-03-24:
However, when the code has been modified, just make linkPin and pvVol become integer declarations, problem solved! Anyone tell me is this a compiler's bug or not?
[image]

matthijskooijman, 2015-03-24:
Is there any error _before_ the ld error? Could you try again with verbose compilation enabled, and post the full output (from the bottom window) and your full sketch code?

laisan86, 2015-03-25:
No! No compilation errors, I checked many times.

ffissore, 2015-03-31:
Can you share the whole sketch?

agdl, 2016-07-25:
https://github.com/arduino/Arduino/issues/1469

closed 2016-07-25 by agdl

HiTechHiTouch, 2016-09-09:
Please see [Issue 2989](https://github.com/arduino/Arduino/issues/2989#issuecomment-245832814).  There are full dumps and source all zipped up, including the Windbg analysis.  There should be more than enough information for someone with the LD.exe source to trace the problem.
[image]