arduino/Arduino #2816: class SPIClass has no member named beginTransaction

adrian-marina, 2015-03-24:
I've upgraded my Arduino IDE to the latest version 1.6.1 for Mac OS X and I cannot compile my sketch anymore due to some errors inside Ethernet library, specifically when it uses the SPI.beginTransaction(SPI_ETHERNET_SETTINGS); apparently class SPIClass has no member named 'beginTransaction'. 
for the purpose of this demonstration I have created a dummy sketch, here is a screen-shot
[image]

I've had a look inside the Ethernet library and found that SPI.beginTransaction(SPI_ETHERNET_SETTINGS); was added in this commit  https://github.com/arduino/Arduino/commit/8aaca2fbb662b293f7089f1505e4a8e922e49b0d

as a quick fix I just went through the files and removed all references to beginTransaction and endTransaction and I can confirm that the sketch now compiles successfully.

can you consider this bug in your next release ? or provide a patch ?

matthijskooijman, 2015-03-24:
The SPI library shipped with 1.6.x should have those functions. Could it b e that there is an old version of the SPI library somewhere in your sketchbook? If not, could your post the full verbose compilation output here?

ffissore, 2015-03-24:
Indeed, 1.6.1 compiles your sketch fine. Since the log shows you're using bundled Ethernet, the only possible source of error is an old SPI library in your sketchbook

PaulStoffregen, 2015-03-24:
Indeed, this is almost certainly an old copy of SPI in the sketchbook libraries.

I've been seeing more and more of this on our forums, where users upgrade to a newer Arduino and are unaware they have outdated copies of libraries in their sketchbook libraries.

So far, I haven't thought of any good way to help them see why things aren't working.

adrian-marina, 2015-03-24:
Well, yes that is the case. I had an old SPI library sitting around in the sketchbook libraries. Being a beginner with Arduino I didn't think about this case.

thanks for your quick help.

closed 2015-03-24 by adrian-marina

PaulStoffregen, 2015-03-24:
We definitely need to improve Arduino to detect this sort of situation and show a useful message that novices can easily understand.

JesusIslam, 2017-01-06:
I am using Arduino IDE v2.1.5 and this still happened.

dhananjay32, 2017-02-24:
Plese share the download link arduini 1.6.1