arduino/Arduino #2817: Arduino Wifi Shield library unable to obtain entire socket buffer.

josephlm, 2015-03-25:
When an incoming UDP packet exceeds > ~90 bytes, the entire buffer does not get passed up through the SPI header because it exceeds some SPI max buffer size. It then appears the packet is never received on the receiver's side in the application layer above the Arduino Wifi Shield Library.

facchinm, 2015-03-25:
Hi @josephlm,
could you post your sketch and the IDE version you are using? 
I can't find any explicit RX buffer in current WiFi library (the one shipped with the IDE 1.6.1 doesn't use the standard SPI library though)

josephlm, 2015-03-25:
I believe I've tracked it down. Please tell me if I'm wrong here, but here goes:

In the firmware for the Arduino WiFi shield, it defines a max buffer size of 100 here:

arduino-1.6.1/hardware/arduino/avr/firmwares/wifishield/wifiHD/src/ard_spi.c at line 78.

I believe that the IP Header info + a data buffer at about ~92 bytes will exceed this limit. The Wifi Shield library consumes this buffer without knowledge of this max buffer size of 100.

facchinm, 2015-03-25:
Yep, you are right! So the culprit is the shield's firmware.
You can try to compile a modified version with larger buffer using atmel studio and upload it following the instructions (http://arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading) 

In the meanwhile I'm closing the issue but feel free to report any result here

closed 2015-03-25 by facchinm

josephlm, 2015-03-25:
It appears that the firmware has different licensing. I think if this should be an issue, the proper fix would be to implement some sort of block transfer mechanism into existing APIs. Otherwise, simply raising that max buffer size for the next firmware version would be best. I'm not familiar with the resource constraints of the Wifi Shield, however, so REALLY know if that is appropriate.

facchinm, 2015-03-25:
Honestly I can see no feasible way to implement a block transfer using the existing API between Arduino and the shield, as every method involves modifying the firmware.

Also, I really don't know if there is any resource constraint but if you wish to spend some time recompiling, flashing and reporting the results it would be great! We could then include your patch in the next fw release.

From the licence side, I think there is no problem at all in modifying and redistributing the code as both Atmel's and H&D Wireless' licences seem very permissive

josephlm, 2015-04-01:
@facchinm I will certainly provide a firmware pacth for the community. After I am finished, How do I go about uploading it? How do we go about the code review process?

facchinm, 2015-04-01:
The source code for the shield's firmware is included in the standard Arduino repo, so the code review process will happen here :smile: 

About uploading, it would be great if you could provide both the patch and a binary firmware replacing the current one or in addition, you decide (we can't obviously force anyone to apply it but it's nice to have it in place)