arduino/Arduino #2818: Software Serial missing

Danyc0, 2015-03-25:
Software Serial seems to have been removed from the new versions of the IDE, i heard it was meant to be replaced by a newSoftSerial library but that doesn't seem to exist either?

hashashin, 2015-03-25:
Hi @Danyc0, for me the library is still there, which IDE version you've?
https://github.com/arduino/Arduino/tree/1.6.1/hardware/arduino/avr/libraries/SoftwareSerial
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/SoftwareSerial

PaulStoffregen, 2015-03-25:
Maybe you have the wrong board selected?

The new IDE doesn't show libraries for AVR when you have a Due selected, for example.  Or if you've installed a non-Arduino board (and selected it in Tools > Boards), maybe Arduino is not picking up that it supports SoftwareSerial.

Danyc0, 2015-03-26:
Yea, that was me being stupid, thanks guys!

closed 2015-03-26 by Danyc0