arduino/Arduino #2822: When there is only one board connected, auto-detect what port it is on

hichris1234, 2015-03-26:
I plugged in my board today on a new machine, and it was on COM3. I'm assuming the default for the IDE is COM1. When I tried to upload my sketch to the board, it hung as COM1 had no available Arduino on it.

Now, this is probably a bug and a feature request:
- bug: the uploading process didn't fail out when there wasn't a board on COM1. It simply hung. This should probably fail out with an error "please check your port... blah blah blah.".
- feature request: if there is no Arduino connected on the default port (COM1), but there is an Arduino connected on a different port, select the port with the Arduino connected to it. This should probably be limited in some way as if someone had two devices connected and deconnected one, it would automatically switch the port and thus cause some annoyance.

Avamander, 2015-05-26:
Some computers (like mine) have RS232 ports that appear available all the time. So it has to communicate with the arduino to determine it can be actually programmed.

ffissore, 2015-05-29:
Such an automatism may solve a problem for some, and cause headaches to others. Wontfix

closed 2015-05-29 by ffissore