arduino/Arduino #2826: Leonardo: Serial.available() always returns 0 or 1

sudden6, 2015-03-27:
On the Arduino Leonardo Serial.available() never returns values greater 1, even if there are more chars in the buffer.

Example:
```
void setup(void) {
  
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }
}

void loop(void) {
  Serial.println(Serial.available());
  delay(500);
}
```

Result: The Leonardo sends 0 as long as there's no character sent and 1 if you send any amount of characters.

Expected result: For every character you send, the value sent from the Leonardo should increase.

facchinm, 2015-03-30:
Hi @sudden6 ,
I've tested your code with a Leonardo and the result is correct (hence writing 8 characters into the textbox of the serial monitor returns 9 ( 8 + end_of_line )

Could you give more info?
If you change your code this way
```arduino
void loop() {
  int rx = Serial.available();
  Serial.println(rx);
  if (rx > 0)
    Serial.readString();
  delay(500);
}
```
does the result change?

sudden6, 2015-03-30:
Hi,
I tested your code and this seems to be a little more complicated. First, your code and my original code both work, but ONLY with the serial monitor integrated into the Arduino IDE.

The other terminal programs I tested: CuteCom (KDE terminal program), putty, and minicom show the behavior I described above.

Can this be some OS specific problem?
I'm running OpenSUSE Tumbleweed with Linux 3.19.0-2-desktop

Or is the Serial Monitor doing something special that other programs don't do?

matthijskooijman, 2015-03-30:
The serial monitor in Arduino queues up all characters in the input box, and then sends them all quickly when you press enter. Most other serial programs just send the characters as soon as you type them. The example by @facchinm should therefore show a difference between the serial monitor and other programs, because the 500ms delay is long enough for the characters from the serial monitor to queue up in the Arduino, but not long enough to have a lot of characters queue up when you have to type them.

The original program should not show any differences - no characters are ever read from the buffer, show `Serial.available()` should just increase whenever enter is pressed (serial monitor) or a character is typed (other programs).

Perhaps this helps clarify things a bit? I don't have a Leo at hand to test, unfortunately.

sudden6, 2015-03-30:
I know that atleast CuteCom queues up all the characters too.

facchinm, 2015-07-13:
Well, I believe we can close this as the behaviour is due to the serial terminal settings and unrelated with AVR CDC code. Reopen if needed.

closed 2015-07-13 by facchinm