arduino/Arduino #2827: IDE 1.6.1 Issues Sloooowww and buggy.

Bluhair, 2015-03-28:
I recently upgraded to IDE 1.6.1 which was working fine.
Windows 7 Ultimate
Newest Java Installed.
Latest Windows updates installed.
--------
IDE 1.6.1  is very slow. Slow initial opening over 70 seconds, slow loading sketches, slow compiling sketches.. Tool menu button takes over 30 seconds, sometimes freezes or locks. Serial monitor slow to open.
------
Help menu Reference is very slow to open then reports 
C:\Program Files\Arduino\reference\arduino.cc\en\Reference\HomePage.html
Internet exploer restricted this webpage from running scripts or Active X Controls
Web page is not responding
Long running Script usually in endless loop.
Verifying and compiling scripts very slow.
Previous version Highlighted errors quickly. IDE 1.6.1 appears to wait until almost finished compiling before reporting error and NO highlighting of errors.
I have installed the IDE 1.6.1  three times. Each time I make sure I have the latest Java and windows updates.
Possibly I missed something or made an error.
I have not see problems like this while searching GOOGLE.

Thank you in advance for any suggestions or answer.

Bluhair, 2015-03-28:
I hope there is a quick and simple fix. Have not found it yet.

PaulStoffregen, 2015-03-28:
Did you try using the nightly build?  A java memory setting was increased in 1.6.1, but it turned out to be too much for some computers.  It was lowered recently, but not back to the too-low setting of 1.6.0.

Bluhair, 2015-03-30:
Thank you Paul.
I tried the nightly build and it solve some of the problems, it was much faster and seemed to work much better.
To complete the test, I need to install the nightly build.
Not sure how to do that. There was no install script or files in the nightly build.
Thank you again.
Charles Cook

Bluhair, 2015-03-30:
Opps did not mean to close this one.

ffissore, 2015-03-30:
Tools menu has much improved since 1.6.2. You may consider closing this anyway

Bluhair, 2015-03-30:
I would like to know ...
How to install the nightly build ?
Or are the nightly builds incorporated into the installable version of the main download?
Not sure how to install.
Thank you

PaulStoffregen, 2015-03-30:
Go to the main download page, the same way as you would to install the normal version.

But instead of clicking in the normal place, simply scroll down the page a little more, to the "Nightly Builds" section.  There, you'll find links to the newer nightly build copies.  Those links work exactly the same as the normal download (except they don't ask you to donate).

Well, the one other difference is there are 4 links instead of 5.  The Windows version is only the ZIP file.  If you've only ever used the Windows EXE installer, you're going to have to learn how to install the ZIP file.  Fortunately, it's easy.  Just extract it to any location on your computer.  It will create a folder with a new copy of Arduino inside.  To run it, open that folder and double click the copy of Arduino.  if you've only ever used the Windows EXE installer, you've probably always run Arduino by double clicking the Arduino icon it put on your desktop.  That icon will still work and still run the original copy of Arduino that came from the installer.  Just remember you now have 2 copies of Arduino on your computer.  Microsoft Windows tries to make things "easier" by giving you defaults and not always showing details of where things really are on your computer.  If you're accustomed to that way of doing things, you're going to have to rise to the challenge of remembering you've got 2 separate copies.  There are plenty of on-screen cues to help you, like the version printed in the title bar of the window and Help > About.

Bluhair, 2015-03-31:
Clarification.
I upgraded FROM Arduino version 1.6.1 (was working fine) 
TO Arduino version 1.6.2 (numerous bugs)
I downloaded the Nightly Build for 1.6.2 and it seemed to work Good.
I want to get a GOOD working version of version 1.6.2 INSTALLED. 
I tried again to download the version 1.6.2 from the Main download. Again not working good.
I have no idea if and when the daily builds are incorporated into the Main download.
My question is...
How can I install a GOOD version 1.6.2 in the correct directory and not have to use a separate sub directory which is crude and not satisfactory.
I hope this clarifies my original question.
Thank you
Charles Cook

ffissore, 2015-03-31:
@Bluhair when the IDE available with the nightly builds will become a release, you'll see version 1.6.3 on the download page. That will happen soon.

ffissore, 2015-04-02:
@Bluhair version 1.6.3 is out. Check it out. I'm closing this as solved. Reopen if you find tools is still slow

closed 2015-04-02 by ffissore