arduino/Arduino #2828: 1.6.2 JNA conflict

rodrigopitanga, 2015-03-28:
Arduino 1.6.2 will crash after splash screen. Looks like it requires a non-standard library.

java.lang.UnsatisfiedLinkError: com.sun.jna.Native.pointerSize()I
	at com.sun.jna.Native.pointerSize(Native Method)
	at com.sun.jna.Native.<clinit>(Native.java:93)
	at cc.arduino.os.linux.LibCNative.<clinit>(LibCNative.java:38)
	at cc.arduino.os.linux.LinuxFileNativeUtils.<clinit>(LinuxFileNativeUtils.java:36)
	at cc.arduino.os.FileNativeUtils.chmod(FileNativeUtils.java:51)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:236)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:585)
	at processing.app.Base.<init>(Base.java:256)
	at processing.app.Base.guardedMain(Base.java:198)
	at processing.app.Base.main(Base.java:113)

Operating System: Trisquel GNU/Linux 7.0 (x86_64)

Java: java version "1.7.0_75"
OpenJDK Runtime Environment (IcedTea 2.5.4) (7u75-2.5.4-1~trusty1)
OpenJDK 64-Bit Server VM (build 24.75-b04, mixed mode)

sej7278, 2015-03-28:
see also #2830 and #2829, think its a missing hardware directory not a java library issue?

aveekbh, 2015-03-28:
Looks like the function `com.sun.jna.Native.pointerSize(Native Method)` is not available. I could not find any reference to `pointerSize` in https://github.com/twall/jna. Unfortunately, I don't know enough Java to identify the issue further. But, yes, the hardware directory is empty.

closed 2015-03-30 by ffissore

ffissore, 2015-03-30:
Fixed with b65b576eb026c6ee5718a8262d9c2b274f6639c3. A nightly build is on the way

commit 1dfede77 (2015-03-30): Getting ready on native stuff. Avoid native stuff: it will break. Fixes #2828, #2829, #2830
commit b65b576e (2015-03-30): Getting rid of native stuff. Avoid native stuff: it will break. Fixes #2828, #2829, #2830