arduino/Arduino #2829: Arduino 1.6.2 fails to start on Linux x64

aveekbh, 2015-03-28:
Arduino 1.6.2 IDE does not start on Ubuntu 14.10 (x64). The splash screen comes up briefly, but there is a Java error:

arduino-1.6.2$ ./arduino 
java.lang.UnsatisfiedLinkError: com.sun.jna.Native.pointerSize()I
        at com.sun.jna.Native.pointerSize(Native Method)
        at com.sun.jna.Native.<clinit>(Native.java:93)
        at cc.arduino.os.linux.LibCNative.<clinit>(LibCNative.java:38)
        at cc.arduino.os.linux.LinuxFileNativeUtils.<clinit>(LinuxFileNativeUtils.java:36)
        at cc.arduino.os.FileNativeUtils.chmod(FileNativeUtils.java:51)
        at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:236)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:585)
        at processing.app.Base.<init>(Base.java:256)
        at processing.app.Base.guardedMain(Base.java:198)
        at processing.app.Base.main(Base.java:113)

The Java version is:
$ java -version
java version "1.6.0_34"
OpenJDK Runtime Environment (IcedTea6 1.13.6) (6b34-1.13.6-1ubuntu0.14.10.1)
OpenJDK 64-Bit Server VM (build 23.25-b01, mixed mode)

sej7278, 2015-03-28:
seeing the same in issue #2830 on debian sid/jessie

```
java -version
java version "1.7.0_75"
OpenJDK Runtime Environment (IcedTea 2.5.4) (7u75-2.5.4-3)
OpenJDK 64-Bit Server VM (build 24.75-b04, mixed mode)
```

i believe its partially (?) due to the tarball containing an empty hardware directory.

mithat, 2015-03-28:
The problem exists for Linux 32-bit versions as well.

~~~~
java.lang.UnsatisfiedLinkError: com.sun.jna.Native.pointerSize()I
	at com.sun.jna.Native.pointerSize(Native Method)
	at com.sun.jna.Native.<clinit>(Native.java:93)
	at cc.arduino.os.linux.LibCNative.<clinit>(LibCNative.java:38)
	at cc.arduino.os.linux.LinuxFileNativeUtils.<clinit>(LinuxFileNativeUtils.java:36)
	at cc.arduino.os.FileNativeUtils.chmod(FileNativeUtils.java:51)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:236)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:585)
	at processing.app.Base.<init>(Base.java:256)
	at processing.app.Base.guardedMain(Base.java:198)
	at processing.app.Base.main(Base.java:113)
~~~~

~~~~
java -version
java version "1.7.0_75"
OpenJDK Runtime Environment (IcedTea 2.5.4) (7u75-2.5.4-3)
OpenJDK Server VM (build 24.75-b04, mixed mode)
~~~~

aptosid Linux (Debian sid derivative)

jan-r, 2015-03-29:
Same problem on Mint 16 64-bit:

java version "1.7.0_55"
OpenJDK Runtime Environment (IcedTea 2.4.7) (7u55-2.4.7-1ubuntu1~0.13.10.1)
OpenJDK 64-Bit Server VM (build 24.51-b03, mixed mode)

Linux 3.11.0-12-generic #19-Ubuntu SMP Wed Oct 9 16:20:46 UTC 2013 x86_64 x86_64 x86_64 GNU/Linux

jmd1979, 2015-03-29:
I have the same problem on ubuntu 12.04 64 bit

java version "1.7.0_75"
OpenJDK Runtime Environment (IcedTea 2.5.4) (7u75-2.5.4-1~precise1)
OpenJDK 64-Bit Server VM (build 24.75-b04, mixed mode)

java.lang.UnsatisfiedLinkError: com.sun.jna.Native.pointerSize()I
	at com.sun.jna.Native.pointerSize(Native Method)
	at com.sun.jna.Native.<clinit>(Native.java:93)
	at cc.arduino.os.linux.LibCNative.<clinit>(LibCNative.java:38)
	at cc.arduino.os.linux.LinuxFileNativeUtils.<clinit>(LinuxFileNativeUtils.java:36)
	at cc.arduino.os.FileNativeUtils.chmod(FileNativeUtils.java:51)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:236)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:585)
	at processing.app.Base.<init>(Base.java:256)
	at processing.app.Base.guardedMain(Base.java:198)
	at processing.app.Base.main(Base.java:113)

q2dg, 2015-03-29:
In my Fedora 21 works well:

openjdk version "1.8.0_40"
OpenJDK Runtime Environment (build 1.8.0_40-b25)
OpenJDK 64-Bit Server VM (build 25.40-b25, mixed mode)

It seems it's a issue with Java version...

P.D: Library and board manager, however, appear without the closing button "X"...mmm

mithat, 2015-03-29:
@q2dg Tried OpenJDK 1.8 on 32-bit aptosid (Debian sid derivative) and got the same errors:

~~~~
java -version
openjdk version "1.8.0_40-internal"
OpenJDK Runtime Environment (build 1.8.0_40-internal-b27)
OpenJDK Server VM (build 25.40-b25, mixed mode)
~~~~

~~~~
java.lang.UnsatisfiedLinkError: com.sun.jna.Native.pointerSize()I
	at com.sun.jna.Native.pointerSize(Native Method)
	at com.sun.jna.Native.<clinit>(Native.java:93)
	at cc.arduino.os.linux.LibCNative.<clinit>(LibCNative.java:38)
	at cc.arduino.os.linux.LinuxFileNativeUtils.<clinit>(LinuxFileNativeUtils.java:36)
	at cc.arduino.os.FileNativeUtils.chmod(FileNativeUtils.java:51)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:236)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:585)
	at processing.app.Base.<init>(Base.java:256)
	at processing.app.Base.guardedMain(Base.java:198)
	at processing.app.Base.main(Base.java:113)
~~~~

sej7278, 2015-03-29:
me too, java 1.8 on 64-bit debian sid gives exactly the same error.

Robot-Will, 2015-03-30:
Google jna-3.2.7.jar and downlad the file, rename it to jna.jar, then replace the same file in <arduino dir>/lib, and Arduino IDE will start.

ffissore, 2015-03-30:
Fixed with b65b576eb026c6ee5718a8262d9c2b274f6639c3. A nightly build is on the way

ffissore, 2015-03-30:
Nightly ready

closed 2015-03-30 by ffissore

aveekbh, 2015-03-30:
Hello Federico

The nightly build from 2015-03-30 starts up fine, but won't compile any sketches. Opened issue #2849

Nzaga, 2016-05-20:
Use **sudo update-alternatives --config java** to check for the jdk version you are using. Change it to 1.7.* . Then start again the arduino