arduino/Arduino #2832: Add Souliss library to the Library Manager

plinioseniore, 2015-03-28:
Hi,

could you please inform on how get the library included and how updates are managed? Is it linked directly with Github tags?

Thanks,
Dario.

testato, 2015-03-29:
Also I am interested, I would insert 3 libraries

ffissore, 2015-04-07:
Sorry for having delayed this one. Please have a look at this [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) and this [email on dev list](https://groups.google.com/a/arduino.cc/d/msg/developers/EK1F2YLoU_U/jTrX5It-waoJ)

If you wish Souliss lib to get listed, ensure it's a valid 1.5 lib and provide its git repo url

plinioseniore, 2015-04-07:
Hi Federico,

is there a way to add the library.properties while maintaining the 1.0.x format? For big libraries like Souliss, where there are several folders, subfolders and files move all the code in the /src one isn't a really suitable solution.

Thanks,
Dario.

PaulStoffregen, 2015-04-07:
Yes, you can add library.properties to a 1.0.x style library.  Simply do NOT create a "src" folder.  If there is no "src" folder, Arduino will compile files from the main folder and "utility" folder.

plinioseniore, 2015-04-07:
@ffissore, I've pushed a new tag with the code updated to support the library manager, please add souliss from the following url https://github.com/souliss/souliss 

@PaulStoffregen, thanks for your suggestion, is working fine.

ffissore, 2015-04-08:
Added. Thanks!

closed 2015-04-08 by ffissore