arduino/Arduino #2833: 1.6.2 Linux 32-bit is missing content of hardware directory

mikaelpatel, 2015-03-28:
Release 1.6.2 for Linux 32-bit does not contain any files in the hardware directory. Something must have gone wrong in the delivery processes.

sej7278, 2015-03-28:
same as #2830, #2829 and #2828 except they're 64-bit

Wackerbarth, 2015-03-28:
The files seem to have moved -- On MacOSX, the board file is now 
```
~/Library/Arduino15/packages/arduino/hardware/avr/1.6.2/boards.txt
```

The compilers and avrdude have also moved. And the method of referencing them (in platform.txt) has changed. For example, the compiler path is now
```
compiler.path={runtime.tools.avr-gcc.path}/bin/
```

This is a MAJOR change which appeared without warning. It "broke" all of the custom hardware specifications that I had installed.
Why wasn't this change "previewed" by ArduinoBot as has occurred with other changes?

sej7278, 2015-03-28:
@Wackerbarth i can't find any boards.txt in the linux tarball at all.

i hope this is a packaging screw-up and they've haven't done something as major as moving all the directories around like that in a minor point-release!

this is the equivalent of the 1.0 to 1.5 move and they just slipped it in under the radar, wtf?!

PaulStoffregen, 2015-03-28:
They changed everything to a package system that installs it all into your ~/.arduino15 folder.

123 commits, developed over a period of several months, where all pushed to this public repository yesterday.  I'm seeing all this code for the first time too....

sej7278, 2015-03-28:
@PaulStoffregen crazy behaviour, i hope it was a screw-up that'll be reverted.

not only does it mess things up for people making supporting software like cores and makefiles, or even hardware, but people writing books and tutorials, and also users.

time for a fork methinks.

PaulStoffregen, 2015-03-28:
One man's "screw-up" is another man's "progress".  :-)

Historically, about every 2 years Arduino has pretty substantially restructured how files and folders are arranged.  Each time, the complexity increases.  If you consider the beginning of 1.5.x was about 2 years ago, and the move from 0017-0019 (multiple "targets" which became "plaforms") was a couple years before that, we're right about due for another major restructuring.

sej7278, 2015-03-28:
seems weird though that we've only just come out of beta, then they just release something as equally complicated in a point release of the stable tree. i'd have though these sort of changes should be in a 1.7 beta development tree.

i'm going to change my makefile to just trawl the entire filesystem looking for random directories where they could have bunged cores/libraries ;-)

mikaelpatel, 2015-03-29:
I figured out how to at least get this working for Cosa. The gcc-avr tools are in the dist directory in default_package.tar.gz. A few mkdir and mv and the structure is back to normal and I can compile Cosa with both the Arduino IDE and Cosa command line build (make). 

BW: There is no info in revisions.txt. There is no discussion of this packaging change on developers. There is no install guide-line.

closed 2015-03-29 by mikaelpatel