arduino/Arduino #2834: IDE not obey Maximized setting when start

walkin-corpse, 2015-03-28:
Just rightclick on Arduino IDE shotcut, choose Properties and you'll see it at Run field. This is Windows' startup option. Default is Normal window but we can choose Maximized or Minimized too. I'd like to start with maximized editor but I failed 'coz IDE handles it improperly - it maximizes splash screen!

Copied from here: http://forum.arduino.cc/index.php?topic=311381

walkin-corpse, 2015-04-05:
Bug _nealry_ remains in 1.6.3. Splash screen now not maximized but IDE still  not obey startup option. I told about MAXIMIZED, but MINIMZED has same problem. IDE just ignores this setting. Bad behavior!

ffissore, 2015-09-17:
Launch4j is missing such feature. See http://sourceforge.net/p/launch4j/feature-requests/78/