arduino/Arduino #2835: Mac build fails with "MacOSX requires MACOSX_BUNDLED_JVM property set"

PaulStoffregen, 2015-03-28:
Does Arduino now require extra setup to build on Macintosh?

PaulStoffregen, 2015-03-28:
BUILD FAILED
    /Users/paul/1.6.2/build/build.xml:58: The following error occurred while executing this line:
    /Users/paul/1.6.2/build/build.xml:258: MacOSX requires MACOSX_BUNDLED_JVM property set

Wackerbarth, 2015-03-29:
I ran into the same problem. Commit b48e07c switched to using the branch that was created to upgrade to a newer version of java. However, although I have the latest JDK and JRE installed, I cannot figure out where I should have this environment variable point. Does anyone know what we need?

PaulStoffregen, 2015-03-31:
It's fixed  https://github.com/arduino/Arduino/commit/319f1c3b632bb52a5814337df9d984618f068db4

closed 2015-03-31 by PaulStoffregen