arduino/Arduino #2838: IDE 1.6.2 breaks semantic versioning standards

Wackerbarth, 2015-03-29:
The changes introduced in 1.6.2 need to be rolled back to maintain compatibility with 1.6.1 and the new changes be introduced on a 1.7 branch.

See http://semver.org/ which, interestingly enough, was referenced in the development of this change.

sej7278, 2015-03-29:
:+1: 

also would be nice if the 1.7 branch used a ~/.arduino17 directory instead of ~/.arduino15, so it could co-exist with 1.5.x - 1.6.1 like we do with 1.0.x

also putting the version of the ide in the library directory structure seems a bit silly too.

ffissore, 2015-03-30:
1.6.2 never meant to break backwards compatibility. But it did and we are working to fix in the coming days.

matthijskooijman, 2015-03-30:
@ffissore  If 1.6.2 causes a compatibility problem, wouldn't it be better to pull the download off the site, or at least put a notice up and recommend users use 1.6.1 for now?

ffissore, 2015-03-30:
@matthijskooijman we are going to release a fix very soon. the problem arises when using custom cores and should be fixed with the current nightly (I've started a build right now, it should be ready in an hour)

sej7278, 2015-03-30:
i don't think automatic rewriting of cores is a good idea if that what your fix is in #ecde17b 

i'd say stick with what @matthijskooijman says and pull the download, make 1.6.2 into 1.7.0 beta and give people some time to fix their sketches, cores, makefiles etc.

tolson2000, 2015-03-31:
I agree with @sej7278 and @matthijskooijman. This big of change should be 1.7.0 beta.

ffissore, 2015-03-31:
Latest fix is available as a nightly build.
Cores using paths like `{runtime.ide.path}/hardware/tools/avr/bin/` or `{runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/` are automatically converted (that happens in-memory, no file is changed). [Here is the list](https://github.com/arduino/Arduino/blob/master/hardware/platform.keys.rewrite.txt) of property values that get rewritten.
Hardware folder content has been restored and it contains the AVR bundle.
I've used [sparkfun core](https://github.com/sparkfun/Arduino_Boards) and a core nicohood submitted a couple of weeks ago and they both work fine

damellis, 2015-03-31:
Does bundling the AVR core mean that it can't be upgraded using the boards
manager?

ffissore, 2015-04-02:
Sorry @damellis , answered privately and missed to paste it here: of course no, bundled core is just as upgradable as any other later installed core

ffissore, 2015-04-02:
1.6.3 is out. Thank you for your feedback. I hope you'll find it more reliable than 1.6.2

closed 2015-04-02 by ffissore