arduino/Arduino #2841: [Feature request] Add to Serial.write() the hability of send byte structs

q2dg, 2015-03-29:
I mean...Serial.write() has already the hability of sending byte arrays in the form Serial.write(myarray, sizeof(myarray));. Why not doing the same with structs like struct { byte x; byte y;} mystruct; ??
Thanks!

matthijskooijman, 2015-03-30:
I guess you already can, by just treating a struct as an array of bytes. I believe something like this should work:

```
struct { byte x; byte y} var;

Serial.write((uint8_t*)&var, sizeof(var));
```

q2dg, 2015-03-31:
You're right, it works!! 
The expression is a little ugly...but I suppose it can't be written in another (more friendly) manner...
Well, it's ok, anyway.
Thanks a lot!!!
 I close the issue.

PaulStoffregen, 2015-03-31:
Serial.write really should have been defined to take a "const void *".  Sadly, it's probably far too late to change now.  Unless someone can think of a crafty way with overloading, so old code overriding Print still works?

q2dg, 2015-03-31:
I reopen the issue to give the opportunity to answer Paul's question