arduino/Arduino #2845: 1.6.2 IDE - support for additional boards - RedbearLab boards ?

henrysinn, 2015-03-30:
Hi.
I've just upgraded to the latest IDE and like / love the idea of a 'boards manager' 
I've followed instructions from http://blog.arduino.cc/2015/03/28/arduino-ide-1-6-2-released/
but unfortunately I'm not seeing the boards I need / have

ie. Some from redbearlab.com

How do I install 3rd party boards in the interim?

ffissore, 2015-03-30:
The usual instructions (sketchbook "hardware" folder etc) will work as usual. We are working on some new cores and libs

henrysinn, 2015-03-30:
Thanks for the note Federico.
Unfortunately it's not. Something isn't working / connecting.. 
Henry

ffissore, 2015-03-30:
It may be because of a bug we introduced in 1.6.2 and that we're fixing.

ffissore, 2015-03-30:
Hold on until we fix it

henrysinn, 2015-03-30:
OK / Thanks for the update

henrysinn, 2015-04-01:
Hi Federico / all.

Any update/s?

Henry

agdl, 2015-04-03:
@henrysinn the 1.6.3 fix the problem. Please try it and close this issue if it works :)

henrysinn, 2015-04-06:
Hi agdl.
Sorry for slow response

Have tried without success...

Pretty easy to try yourself even if  you don't have a board:
1. Get "RedBearLab Blend Add-On version 20140701" from http://redbearlab.com/getting-started-blendmicro
2.  "Follow the yellow brick road" - steps 5, 6, 7, and 8 in particular - the rest doesn't matter 

Henry

sakulstra, 2015-04-07:
Same problem here :+1: 

Regards,
Lukas

cmaglie, 2015-04-07:
Looking at the "source code" link it seems that RedBearLab has released an 1.6.x update:

https://github.com/RedBearLab/Blend/

Please try the archive download from github, here the link: https://github.com/RedBearLab/Blend/archive/master.zip

henrysinn, 2015-04-07:
Hi cmaglie,

This works.

Henry

closed 2015-04-07 by henrysinn