arduino/Arduino #2847: Advise users of duplicate libraries

PaulStoffregen, 2015-03-30:
Sometimes Arduino users will install a duplicate copy of a library, or another library that appears to be a duplicate due to conflicting header file names.  Sometimes this actually works, but breaks months or years later when other code depending on those libraries updates, long after the user as forgotten they installed the library.

Arduino should detect duplicate or ambiguous library matches and print a friendly warning after compiling, even if no error resulted, to give users clear visibility that it had to guess which library to build into their program.

PaulStoffregen, 2015-03-30:
@ffissore - I'm working on a patch with the new 1.6.2 code.

The difficult part is the new LibrariesIndexer class.  As it rescans libraries, it uses LibraryList addOrReplace(lib).  That produces a nice list, but a lot of information about possible duplicates is lost.  I have the rest working, and now I'm trying to extend LibraryList to also create a list with duplicates preserved.  Does that sound ok?  Or would you prefer this done some other way?

ffissore, 2015-03-30:
Sounds ok to me

ffissore, 2015-04-01:
#2850 merged, closing

closed 2015-04-01 by ffissore