arduino/Arduino #2849: Nightly build (fixing issues #2828, #2829, #2830) is not compiling sketches

aveekbh, 2015-03-30:
With the nightly build (2015-03-30), sketches fail to compile - even the bare minimum example. The error is:
```
avr-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
compilation terminated.
Error compiling.
```
Platform: Ubuntu 14.10 x86_64 GNU/Linux
avr-gcc (GCC) 4.8.1

Verbose output added in http://pastebin.com/6DejC6LM

aveekbh, 2015-03-30:
The verbise output in http://pastebin.com/6DejC6LM refers to `/h/a` - this was actually `/home/<username>`, which I edited out before uploading publicly..

aveekbh, 2015-03-30:
Moving my `~/.arduino15` directory to force Arduino to create it afresh made no difference, sketches still don't compile.

The SHA256 sum of the nightly build download `arduino-nightly-linux64.tar.xz` is `f39363dd2426ac11227ef7df5f22a6301c4bf1b417f6e5695f2e6c458ec1ebc7`.

ffissore, 2015-03-30:
Yes, I fixed that in https://github.com/arduino/Arduino/commit/1fcd6ff1cedf64999432d4a86194669b65e62c04. The download server are pretty overloaded so the newer nightly (I manually started) is taking a bit to come up. Just wait a couple of hours

aveekbh, 2015-03-30:
Hello Federico

This showed up in Ubuntu. Perhaps that label should also be added?

ffissore, 2015-03-30:
Right, sorry for having misread. Indeed, the problem is caused by a missing piece of the toolchain. Every toolchain has been updated and the next nightly bundles core 1.6.3, which is identical to 1.6.2 but depends on an updated gcc. Again, just wait a couple of hours. I'm monitoring the CI server: I'll keep you posted when the newest nightly is available

ffissore, 2015-03-30:
Nightly available, cores upgraded

Pitel, 2015-03-31:
It works for me.

(But I'm still affected by #2853 so I still can't upload.)

aveekbh, 2015-03-31:
Looks like it's fixed now. I had to rename my `~/.arduino15` directory to get this to work.

For the record, SHA256 sum of `arduino-nightly-linux64.tar.xz` is `32cbc18f3092ab6b283c49d44d07577e8ca8a294abd3efcbafa52c33c4bced2d`

closed 2015-03-31 by aveekbh