arduino/Arduino #2851: The text in IDE is unreadable on High Resolution screens

punitganshani, 2015-03-30:
On high-resolution screens (3200x1800), the text in IDE is very small.  Even after reducing the resolution to 1920x1080, the text is too small to read.

[image]

PaulStoffregen, 2015-03-30:
This is a duplicate of #2532

See pull request #2776 for a partial solution.  Please test #2776  (don't forget to edit theme.txt) on your computer+display and comment.  Maybe someday it will be merged into Arduino, but so far very few people with these highres screens have tested it.

00alis, 2015-04-03:
Have you tried changing the font size in the Preferences? 
[image]

I am using a Mac retina display 15" and I can see the text pretty big with the font set at 22 pt
[image] 

Of course this does not fix the UI and menu size problems, if you are referring only to that please discard this comment.

neslekkim, 2015-04-21:
Does it exist built versions with this fix to test?, on my 4k display the ide is rather unusable.

PaulStoffregen, 2015-04-21:
Yes, test versions exist on issue #2776.  About half way down that page are test copies from ArduinoBot.

You MUST edit theme.txt.  They don't (yet) scale automatically.  On a Mac, control-click on Arduino and select "Show Package Contents".  Then find "theme.txt" inside and edit to increase the scaling.  In that test version, the first line in theme.txt is the new scaling parameter.

Please comment on issue #2776.  This duplicate issue should be closed.

closed 2015-04-22 by ffissore