arduino/Arduino #2853: Port menu greyed out on Linux

Pitel, 2015-03-30:
Using latest nightly. It works ok in 1.6.1. Ubuntu 14.10 64bit.

aveekbh, 2015-03-31:
Hello Pitel

It worked for me after two steps:
1. Moved my ~/.arduino15 directory (to get the fix for #2849 to work).
2. Selected Yun as my board, then Uno again (the actual board that I tested this on). Then, the port menu showed up.

Pitel, 2015-03-31:
Just downloaded latest nightly, deleted `~/.arduino15`, and it works.

Pitel, 2015-04-02:
It still happens occasionaly, and it's pretty annoying. It starts to working after your workaround, but it's still just a workaround.

ffissore, 2015-04-02:
Can you list the ports you see when that happens? (in order to understand if it's a mix of serial and network or only one of the two)

Pitel, 2015-04-02:
Well, when it happens I don't see any because the whole *Ports* menu is greyed out. But it starts working, I see these items:
* `/dev/ttyACM0`
* `/dev/ttyS`

The `/dev/ttyACM0` is the one that works.

facchinm, 2015-04-14:
Probably a duplicate of #2892 , should be fixed in nightly builds.
Reopen if needed

closed 2015-04-14 by facchinm