arduino/Arduino #2854: How are new libraries added to library_index.json?

ryantm, 2015-03-30:
Hi.

I work at [Pololu](https://www.pololu.com). We have a number of [Arduino libraries](https://github.com/pololu?utf8=%E2%9C%93&query=arduino) that I'm hoping we can include in the new Library Manager introduced in 1.6.2. I see that a [library.properties](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/library.properties) file needs to be added to the  library's root directory to specify the metadata for the library.

How do we submit the libraries for inclusion into http://downloads.arduino.cc/libraries/library_index.json ?

Ryan

ffissore, 2015-03-31:
Hi @ryantm. We still don't have a nice web form to fill with the details: for now, an issue on github with the list of git repos to fetch the code from is fine

toddtreece, 2015-03-31:
@ffissore i've got a couple related questions that I couldn't find answers to:
* How are versions handled? I read somewhere that the plan is to follow semver, but how do I submit a new version once I submitted the URL?
* Is the plan to support one installed library version at a time, and if so how will dependencies eventually work? Here's the scenario I'm thinking of: I wrote library **X v1.0.0** that depends on library **Y v1.1.0**, but library **Y v2.0.0** is currently installed and depended on by another library. Assuming the libraries are following semver, the currently installed **Y v2.0.0** would not work with my lib.
* Are library names first come first serve?

ffissore, 2015-03-31:
(initial note: a faq is in the writing)
* Once the url is in the list handled by the job, you as a library author just need to change your code, increment version in library.properties, push and tag the release on git. In at most an hour, your library will be published
* Lib manager does not handle dependencies: at the moment you can see it as a easier way to install a library compared to manually working with zip files and so on.
* A known limitation is that you can't have 2 libs with the same name, just as you can't have 2 folders with the same name

ffissore, 2015-03-31:
Closing this issue as it's becoming a support thread, that is off topic to github issues

closed 2015-03-31 by ffissore

ryantm, 2015-04-01:
Here is the FAQ https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ