arduino/Arduino #2855: library.properties links do not work if the protocol is https

ryantm, 2015-03-30:
Hi.

I am using Windows 8.1 Pro and Arduino 1.6.2. When I click on a "More info" link in the Library Manager it opens if the link has the HTTP protocol but not if it has HTTPS. For example, clicking on the Adafruit NeoPixel link https://github.com/adafruit/Adafruit_NeoPixel does not do anything, but clicking on the AltSoftSerial link http://www.pjrc.com/teensy/td_libs_AltSoftSerial.html opens the link in Google Chrome. I also verified the same behavior by changing the protocol of the URL in the library.properties file for my own library installed on my machine.

Ryan

ffissore, 2015-03-31:
On my linux box, where firefox is the default, it opens adafruit link properly. I'll do some more tests with other setups

ffissore, 2015-03-31:
One thing comes to mind: the link is not clickable on unselected row. You must first click on a row to select it and only then the link becomes clickable: is this the source of your issue?

ryantm, 2015-03-31:
@ffissore Yes, I noticed that and I selected the row before I tried.

ryantm, 2015-03-31:
I reproduced this problem on another co-worker's computer that is also Windows 8.1 Pro.

ryantm, 2015-03-31:
I confirmed this is not a problem on Ubuntu 14.04
```
Linux 3.13.0-46-generic #75-Ubuntu SMP Tue Feb 10 15:24:04 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
````

ffissore, 2015-04-02:
Found and fixed with https://github.com/arduino/Arduino/commit/3a916633b1e3ac2f54e6886b9693937aa9fcd442 will be available with next nightly

closed 2015-04-02 by ffissore

ryantm, 2015-04-02:
Confirmed that 1.6.3 fixes this on Windows 8.1 Pro.