arduino/Arduino #2857: Arduino 1.6.2 64-bit and nightly does not work on Ubuntu 14.10 (Java error)

lekum, 2015-03-31:
1.6.1 works but 1.6.2 throws this error at startup and no splashscreen is shown:

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: no splashscreen in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1119)
	at java.awt.SplashScreen$1.run(SplashScreen.java:124)
	at java.awt.SplashScreen$1.run(SplashScreen.java:122)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.SplashScreen.getSplashScreen(SplashScreen.java:121)
	at processing.app.Base.<clinit>(Base.java:73)
```

I have also tried with the nightly build as of 31st of march 2015, because similar errors have been reported and supposedly solved in that nightly build, but this error is still present.

My system is:

- Xubuntu 14.10
- openjdk version "1.8.0_40-internal" (OpenJDK Runtime Environment (build 1.8.0_40-internal-b09) OpenJDK 64-Bit Server VM (build 25.40-b13, mixed mode)

Regards,

ffissore, 2015-03-31:
This is a different error from the one reported so far. It looks like the jvm you're using does not support splashscreens. That `internal` makes me think it's kind of a development build: where have you downloaded it from?

lekum, 2015-03-31:
Ok, you're right, the JRE that I had install had, been built `headless`, I didn't noticed it. Installing the default version (not the `headless` one) worked. Sorry for noise, and thanks.

closed 2015-03-31 by lekum

ffissore, 2015-03-31:
:+1: thanks for the heads up