arduino/Arduino #2858: SoftwareSerial returns impossible values

Goz3rr, 2015-03-31:
I have the follow code:

    if(_bt.available() > 0) {
    	int data = _bt.peek();
    	bool found = true;

    	if(data == 'a') {
    		// do something
    	} else if(data == 'b') {
    		// do something else
    	} else {
    		found = false;
    	}

    	if(found) _bt.read();
    }

At random points the Arduino would stop reacting to any received messages, and after adding some debug code I found out that while `_bt.available()` would keep incrementing with each message received, `_bt.peek()` returns some a nonsense negative value but `_bt.read()` will return something completely different, these are some i've encountered so far:

-1 == 255
-76 == 180
-1 == 255
-36 == 220

I'm not sure where these values originate from because I'm never sending them to the HC-05 attached to this SerialPort. Worth noting is that once this happens even if i disconnect the HC-05 the SoftwareSerial will keep returning the last values, so I assume some buffering is done internally.

matthijskooijman, 2015-04-02:
What board are you using? Could you see if you could reproduce this problem using a minimal sketch, preferably without any external hardware attached?

Are you sure you're using the official SoftwareSerial library, and not some other library inside your sketchbook? Perhaps you should post the full verbose compilation output, which should show this.

Goz3rr, 2015-04-02:
These are school supplied Arduino Uno's and HC05 Bluetooth modules, I'm not sure if they're genuine Arduinos but I have managed to reproduce it with multiple Arduinos and HC05's and a minimal sketch, I haven't been able to recreate it without external hardware because it doesn't seem to happen while no data is being sent, however it does seem to happen more often when messages are sent more often, baud rate is 9600 so that shouldn't be too high. I'm currently on a bus but I'll post compilation output in a moment.

matthijskooijman, 2015-04-02:
Cool, thanks!

If you can reproduce it by using output from another Arduino (connect its TX pin to the softwareserial RX pin), that would also help a lot. I don't think genuine or clone Arduino would make much of a difference here, btw.

Goz3rr, 2015-04-08:
So far I haven't been able to reproduce this between two Arduinos, it only seems to happen with this specific bluetooth module

sbrynen, 2015-06-15:
You're looking at the same thing; just the difference between signed and unsigned numbers.
byte signed: -1 = byte unsigned 255

If you look carefully at all the numbers you supplied, you'll notice that the difference between the two numbers is always 256.

sandeepmistry, 2016-03-09:
There is a subtle difference between ```SoftwareSerial::read()``` and ```SoftwareSerial::peek()```, ```read``` converts the buffered byte to a ```uint8_t``` before returning, ```peek``` does not.

I'll prepare a pull request to correct this.

Goz3rr, 2016-03-09:
Cool to see someone bothered to look at old issues, and in retrospect that makes a lot of sense

sandeepmistry, 2016-03-09:
I've submitted #4669 to resolve this.

PR #4669 (2016-03-31): Add missing cast to SoftwareSerial::peek() when returning buffered value
Resolves #2858.

Prior to this, ```read``` and ```peek``` would return different values if the byte value was greater than 127.

For example, if the buffered byte was ```180```, ```read``` would return ```180```, but ```peek``` would return ```-76```.
closed 2016-03-31 by agdl