arduino/Arduino #2859: [Feature request] Add String split() function

q2dg, 2015-03-31:
If you want to control your Arduino circuit sending (via Serial, for instance) a sole string with several values inside (each with diferent meaning), a split() function could be very useful. I mean...something similar to http://www.w3schools.com/jsref/jsref_split.asp 
Thanks

PaulStoffregen, 2015-04-01:
Are you proposing implementing regular expressions on Arduino?

q2dg, 2015-04-01:
Nooo! I'm only suggesting it would be nice to be able to split a String into an array (of Strings or of char arrays) based on the presence of certain ASCII character acting as a separator

sej7278, 2015-04-02:
Neither C nor C++ have a split() function by default, so its unlikely to make it into Arduino, which doesn't run javascript ;-)

seriously though, i don't think the arduino could reliably split a string from the serial terminal anyway, you have to add some sort of checksumming etc. to make sure you got the full string and didn't miss a new transmission whilst processing the current one (bit like tcp/ip).

q2dg, 2015-04-02:
Well, I was thinking of Serial.readString()/Serial.readStringUntil() methods. 
They return a String object, which could be split using a hypotethical sentence like 
*char* myarray[]="returnedString".split('separator');*

q2dg, 2015-04-06:
Well...I'm not alone...
http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__string_1ga6ace85338eafe22a0ff52c00eb9779b8.html

q2dg, 2016-08-10:
I think it could be a nice addition to parsing POST/GET querystring parameters, for instance