arduino/Arduino #2860: Software Serial Timing problem with IDE 1.6.x

ProfEdyB, 2015-03-31:
I'm communicating on a software serial link between an Arduino Uno and a Sparkfun Serial motor controller at 115200 Baud, sending five characters. When I updated the IDE to 1.6.0 the serial motor controller stopped responding to commands. I updated IDE to 1.6.2, and same problem. When I back down to IDE 1.5.0, no problem. Looking at the link on a scope, the IDE 1.5.0 correctly completes the 5 character transmission in approximately 430us.  (5 char @10bits/char(includes start/stop) =50 bits*(1/115200)/bit = 434us).  Using IDE 1.6.x, the transmission takes approximately 470us. This is a total error of about 40us or 10%.  This is a problem!  Has anyone experienced this?

matthijskooijman, 2015-03-31:
Interesting observation. In a recent 1.5.x, a pull request by me was merged that overhauls the delay calculations in SoftwareSerial, since the new toolchain messed up the previous hardcoded values. In my testing, this allowed 115200 baud, which was broken before.

I wonder if anything changed afterwards that broke this again, or if perhaps your motor controller is more picky about timing (or perhaps the link is worse). Could you post a minimal sketch that sends those 5 characters? I'll see if I can use that to reproduce the bad timing here.

ProfEdyB, 2015-03-31:
On review of our code, we are using a library called "SendOnlySoftwareSerial" which is an adaptation of "SoftwareSerial" for unidirectional communication (i.e., it frees up the receive pin). The timing in this code is somehow dependent on which IDE is used, not sure why? I did develop a little sketch to test "SoftwareSerial" and it does send with correct timing at 115200 baud in both the older 1.5.x and version 1.6.x IDE.  Thank you.

closed 2015-03-31 by ProfEdyB

matthijskooijman, 2015-03-31:
Cool!

If that library is based on the official SoftwareSerial library, it probably uses a hardcoded number of loops for the timings. Since some recent 1.5.x version upgraded the gcc toolchain, the generated code became different, so the timing of a single loop changed, breaking the signal.

C2BB, 2019-03-10:
For anyone else that comes across this, looks like 1.6.0 is the last version with the old method.