arduino/Arduino #2867: parseInt documentation issues

matthijskooijman, 2015-04-01:
There are a number of problems with parseInt documentation:
 - There are three places where it is documented. One is about Stream::parseInt and two about Serial::parseInt. All three of them are different, even thought the refer to the same function
     - http://arduino.cc/en/Reference/ParseInt
     - http://arduino.cc/en/Reference/StreamParseInt
     - http://arduino.cc/en/Serial/ParseInt
 - There is a single optional parameter, called `skipChar`, which specifies a non-digit character that is skipped inside the number. This can be used for e.g. a thousands separator.
 - The return type should be `long`, not `int`. Previously, #870 was closed stating that `parseInt` really returns `int`, but the code disagrees: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.cpp#L114
 - The documentation should clearly specify how `parseInt` works. In particular, it should state that:
    - Initial characters that are not digits or a minus sign, are skipped
    - Parsing stops when no characters have been read for a configurable timeout value, or a non-digit is read
    - If no valid digits were read when the timeout occurs, 0 is returned
   Currently, none of the three documentation pages have _all_ of the above info.

This issue is partly based on, and repalces #2154, #1713 and #870

agdl, 2015-04-03:
@matthijskooijman i modified

http://arduino.cc/en/Reference/ParseInt

If you think is ok i will modify the others according to this

matthijskooijman, 2015-04-03:
The skipChar parameter is not documented, and there should probably be a link to how to the timeout can be set (I don't know off-hand).

agdl, 2015-04-03:
do you mean as done here? (timeout)

http://arduino.cc/en/Serial/ReadBytesUntil

matthijskooijman, 2015-04-03:
Yup, though you should probably check if that `setTimeout` is actually used for `parseInt` as well. If so, `setTimeout` documentation should list `parseInt` as well.

One more thing: It seems the font in the `parseInt` page for the list you probably copied from my post, is different from the rest of the page.

agdl, 2015-04-03:
The font is different because I used a list style format to empathise the key points :)

agdl, 2015-04-03:
Should be good now. If you agree with me I'll indicate the same things in the other references

matthijskooijman, 2015-04-03:
It doesn't indicate that skipChar is optional yet...

agdl, 2015-04-03:
I stated in the syntax that it can be written in two ways

matthijskooijman, 2015-04-03:
Oh indeed, missed that. Looks acceptable to me now (though the font issue still seems to be there?)

agdl, 2015-04-03:
Actually is not an issue. In the used language that font is related to the list usage

matthijskooijman, 2015-04-03:
Ah, right. No more comments from me, then :-)

agdl, 2015-04-03:
Updated the other pages

closed 2015-04-03 by agdl