arduino/Arduino #2868: CardInfo example doesn't work with nightly

Pitel, 2015-04-01:
Works with 1.6.1, can't test with 1.6.2 because I'm on Linux (64bit).

When compiled and uploaded with 1.6.1, I get nice info about card and files. On nightly I just get this:
```
Initializing SD card...initialization failed. Things to check:
* is a card is inserted?
* Is your wiring correct?
* did you change the chipSelect pin to match your shield or module?
```

I cleaned `.arduino15` before installing nightly. Nightly's MD5 is `b20bce289c0dc848eee665f81f60b50a`.

Pitel, 2015-04-01:
I also fixed the typos in #2869 :)

facchinm, 2015-04-01:
Hi @Pitel ,
does it work if you add 
```arduino
digitalWrite(10, HIGH)
```
just after 
``` arduino 
pinMode(10, OUTPUT);
``` 
?

It could be related to a change (#2659) that solves a problem for someone but could create a lot of trouble to existing sketches.
I'll wait for your feedback before reverting it :smirk:

Pitel, 2015-04-02:
Yes, it works after adding `pinMode(10, OUTPUT)`.

vicatcu, 2015-04-03:
I've made an update regarding issue #2659 that may satisfy both concerns. See comments in the.

Pitel, 2015-04-04:
Well, I'm quite new to Arduino, so I don't know much about this SPI thing. But my general opinion is against legacy stuff. If there is new better/faster/more correct way of doing somehting, and it's not backward compatible, then use the new way. But! Make the change as visible as possible, so that one can easily find how to adopt this new change in their old project.

facchinm, 2015-04-15:
Fixed in #2974

closed 2015-04-15 by facchinm