arduino/Arduino #2870: New StereoSID library to add to Library Manager

daitangio, 2015-04-01:
Hi, I'd like to add my library to Arduino Library Manager (I have read the FAQ at https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).
The library is StereoSID, a SID emulator with stereo capability.
It is in developement, and works well.
You can find it at https://github.com/daitangio/sid-arduino-lib
Current stable release:
https://github.com/daitangio/sid-arduino-lib/releases/tag/StereoSID-2.0.2

ffissore, 2015-04-07:
library.properties misses 'url' field

daitangio, 2015-04-08:
Sorry, I have just fixed it and tested with IDE 1.6.x.
Do I need to add other file/fields?

ffissore, 2015-04-08:
I think not. Once tagged, if successful, in at most an hour you'll see your lib listed

daitangio, 2015-04-09:
Hi, I just retagged it with a new label and the correct library.properties
https://github.com/daitangio/sid-arduino-lib/tree/StereoSID-2.0.3
Is it right or must I do something else?
Thank you :)

ffissore, 2015-04-09:
It's already in the list of repo. It should come up in the IDE in at most an hour

ffissore, 2015-04-09:
And thank you! :)

closed 2015-04-09 by ffissore