arduino/Arduino #2871: Boards Manager - unable to install Due in 1.6.2

rogerclarkmelbourne, 2015-04-01:
The Tools and Cores for Due seem to have been removed from the packages_index.json file

This means anyone running 1.6.2 will not be able to install the Due

Expecting everyone to keep downloading updates of the whole IDE is really not good practice when the IDE is huge.

I'm not sure if I downloaded Today's version of 1.6.2 this would not be an issue, however I have yesterdays's 1.6.2 and it is (and I suspect anyone who downloaded on Saturday  - Tuesday would also have this issue.

agdl, 2015-04-02:
Which OS do you use? I was able to install it both in win7 and win8

rogerclarkmelbourne, 2015-04-02:
The IDE downloads the package_index.json file every time i open the Boards Manager

I'm not sure why the Due boards had vanished this morning, perhaps it was a download issue on the file

I've retrieved the file it downloaded this morning (i.e around 12 hours ago)
(see below)

It didn't contain the Due, so I was unable to install it

I agree the file now contains the Due sections. 

However, because the IDE is downloading the package_index every time I open the boards manager, I feel out of control of the installation.

I also agree it has not removed the Due, but it no longer gave me the ability to install it.

```
{
  "packages": [
    {
      "name": "arduino",
      "maintainer": "Arduino",
      "websiteURL": "http://www.arduino.cc/",
      "email": "packages@arduino.cc",
      "platforms": [
        {
          "name": "Arduino AVR Boards",
          "architecture": "avr",
          "version": "1.6.2",
          "category": "Arduino",
          "url": "http://arduino.cc/download.php?f=/cores/avr-1.6.2.tar.bz2",
          "archiveFileName": "avr-1.6.2.tar.bz2",
          "checksum": "SHA-256:2909a4c6dd6d7497e7e1b5fcaa2f66a100271417510f3a68593b65af8ff78c1c",
          "size": "4877442",
          "boards": [
            {"name": "Arduino Yún"},
            {"name": "Arduino Uno"},
            {"name": "Arduino Diecimila"},
            {"name": "Arduino Nano"},
            {"name": "Arduino Mega"},
            {"name": "Arduino MegaADK"},
            {"name": "Arduino Leonardo"},
            {"name": "Arduino Micro"},
            {"name": "Arduino Esplora"},
            {"name": "Arduino Mini"},
            {"name": "Arduino Ethernet"},
            {"name": "Arduino Fio"},
            {"name": "Arduino BT"},
            {"name": "Arduino LilyPadUSB"},
            {"name": "Arduino Lilypad"},
            {"name": "Arduino Pro"},
            {"name": "Arduino ATMegaNG"},
            {"name": "Arduino Robot Control"},
            {"name": "Arduino Robot Motor"}
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "4.8.1-arduino2"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.0.1-arduino2"
            }
          ]
        }
      ],
      "tools": [
        {
          "name": "avr-gcc",
          "version": "4.8.1-arduino2",
          "systems": [
            {
              "size": "24443285",
              "checksum": "SHA-256:c19a7526235c364d7f62ec1a993d9b495973ba1813869ccf0241c65905896852",
              "host": "i386-apple-darwin11",
              "archiveFileName": "avr-gcc-4.8.1-arduino2-i386-apple-darwin11.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avr-gcc-4.8.1-arduino2-i386-apple-darwin11.tar.bz2"
            },
            {
              "size": "27152002",
              "checksum": "SHA-256:24a931877bee5f36dc00a88877219a6d2f6a1fb7abb989fd04556b8432d2e14e",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-4.8.1-arduino2-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avr-gcc-4.8.1-arduino2-x86_64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "25876628",
              "checksum": "SHA-256:2d701b4efbc8cec62dc299cde01730c5eebcf23d7e4393db8cf7744a9bf1d3de",
              "host": "i686-linux-gnu",
              "archiveFileName": "avr-gcc-4.8.1-arduino2-i686-pc-linux-gnu.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avr-gcc-4.8.1-arduino2-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "46046691",
              "checksum": "SHA-256:2eafb49fb803fa4d2c32d35e24c0b372fcd520ca0a790fa537a847179e382000",
              "host": "i686-mingw32",
              "archiveFileName": "avr-gcc-4.8.1-arduino2-i686-mingw32.zip",
              "url": "http://arduino.cc/download.php?f=/tools/avr-gcc-4.8.1-arduino2-i686-mingw32.zip"
            }
          ]
        },
        {
          "name": "avrdude",
          "version": "6.0.1-arduino2",
          "systems": [
            {
              "size": "264965",
              "checksum": "SHA-256:71117cce0096dad6c091e2c34eb0b9a3386d3aec7d863d2da733d9e5eac3a6b1",
              "host": "i386-apple-darwin11",
              "archiveFileName": "avrdude-6.0.1-arduino2-i386-apple-darwin11.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avrdude-6.0.1-arduino2-i386-apple-darwin11.tar.bz2"
            },
            {
              "size": "292541",
              "checksum": "SHA-256:2489004d1d98177eaf69796760451f89224007c98b39ebb5577a9a34f51425f1",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avrdude-6.0.1-arduino2-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avrdude-6.0.1-arduino2-x86_64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "283209",
              "checksum": "SHA-256:6f633dd6270ad0d9ef19507bcbf8697b414a15208e4c0f71deec25ef89cdef3f",
              "host": "i686-linux-gnu",
              "archiveFileName": "avrdude-6.0.1-arduino2-i686-pc-linux-gnu.tar.bz2",
              "url": "http://arduino.cc/download.php?f=/tools/avrdude-6.0.1-arduino2-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "241618",
              "checksum": "SHA-256:6c5483800ba753c80893607e30cade8ab77b182808fcc5ea15fa3019c63d76ae",
              "host": "i686-mingw32",
              "archiveFileName": "avrdude-6.0.1-arduino2-i686-mingw32.zip",
              "url": "http://arduino.cc/download.php?f=/tools/avrdude-6.0.1-arduino2-i686-mingw32.zip"
            }
          ]
        }
      ]
    }
  ]
}
```

ffissore, 2015-04-02:
Downloading that file every time is something we need to fix. It's a waste of patience and user bandwidth (although a few tens of Kb, but they are likely to grow). And btw the file doesn't change every minute, so the IDE must download only if it's changed. This is something will improve in 1.6.4

And 1.6.3 bundled package_index file is the latest package_index file available, so the lack of due stuff in the list is fixed

PaulStoffregen, 2015-04-02:
What good will the option to install Due be if your internet connectivity (or other issues anywhere in the chain, including management of the files on Arduino's server) isn't even good enough to get the small json file delivered successfully to your PC?

It may give you a warm fuzzy feeling to see the option, but what good is it if it won't actually work?

ffissore, 2015-04-02:
1.6.3 is out. Thank you for your feedback. I hope you'll find it more reliable than 1.6.2

closed 2015-04-02 by ffissore