arduino/Arduino #2872: Request to add new library: aREST

marcoschwartz, 2015-04-02:
Hello,

I am the maintainer of the aREST (http://arest.io/) library that implements a RESTful API for Arduino. It works on most Arduino boards, and can be used via HTTP (Ethernet or WiFi), Serial, or Bluetooth LE.

It's used by several hundreds of people for their projects, I have about 200 views/day on the library page and I have a lot of interaction with users. The link:

https://github.com/marcoschwartz/aREST

I just updated the library with a library.json & library.properties files. Could you add it to the Arduino library manager? If you need more details let me know. Thanks!

Cheers,
Marco Schwartz

ffissore, 2015-04-07:
There are no tags on lib repo. Please review https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

marcoschwartz, 2015-04-08:
I just added a release today with a tag. Let me know if this is ok now, thanks.

ffissore, 2015-04-08:
Added. Thanks!

closed 2015-04-08 by ffissore