arduino/Arduino #2878: No way to remove library from the library manager dialog.

damellis, 2015-04-02:
I just installed the Adafruit CC3000 library using the library manager dialog in Arduino 1.6.3. No there's no way to remove the library from the dialog. Shouldn't this be possible?

ffissore, 2015-04-02:
I've written a short explanation: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-delete-a-library Does it make sense?

damellis, 2015-04-02:
Kind of...

I understand why you wouldn't want to delete libraries that the user was developing or editing. On the other hand, it seems confusing that I can click a button to install a library but can't click another button to remove it. Maybe there should be a check to see if the library has been edited since it was installed and, if so, you could give a warning before deleting it? 

For libraries that were installed manually, it doesn't seem so strange to require them to be deleted manually. (It might be convenient to be able to delete them from the dialog, but I understand why you might not want it.)

Also, is it possible to move libraries into the user's trash instead of deleting the files outright? That might eliminate some of the concern about having a simple way to remove libraries.

ffissore, 2015-04-02:
Accessing trashbin could really help: I've never investigated how to do that. Anyway, you seem to have my same point of view (allow deletion of "managed" libs, not user libs), but it clashes with that of Alice :)

It's a UX issue we should discuss on basecamp

q2dg, 2015-04-02:
Moving files to trashbin could be achieved using these (bundled?) utilities:

-Linux: There is gvfs-trash (only for Gnome) and some utilities: trash-cli (requires Python), xdg-trash (requires Node)...
-Windows: http://www.maddogsw.com/cmdutils or https://github.com/kizzx2/cmd-recycle
-OS X: https://github.com/reklis/recycle or https://github.com/sindresorhus/osx-trash or http://hasseg.org/trash

ffissore, 2015-04-02:
Thanks @q2dg , I'll loot at them

wayoda, 2015-04-04:
> Also, is it possible to move libraries into the user's trash instead of deleting the files outright? 
No worth the effort! When the user finds out later it was an error to delete a library he can always download and reinstall it from the library manager. 
When the user starts to restore a library from the trashbin there is a good chance to mess up the arduino library dir.

PaulStoffregen, 2015-04-04:
Perhaps the IDE should compare all of a library's files to their originals, before deleting.  If any are modified, maybe a "are you sure you want to discard your changes" dialog could appear?  Or maybe the IDE should refuse to delete and suggest the user has to do it manually.

TwatBurglar, 2016-01-28:
ffissore, I disagree with your logic on specifically excluding a delete function.   I do not want nor need your "help" in preventing me accidentally deleting stuff.

What I *do* want is to uninstall all the uneeded crap and clutter out of the library manager.

lmihalkovic, 2016-01-29:
I suggested the idea of leaving libs installed via mgr in their uncompressed form (using them as is), which would assist with the assessing if it was modified (zipped=can delete), as well as allow multiple versions to dit in the libraries folder side by side without any risk of interference.

ffissore, 2016-01-29:
@TwatBurglar I'm sorry you've missed the [news](https://groups.google.com/a/arduino.cc/forum/?fromgroups#!topic/developers/YJLX6AZHem4). I no longer work for Arduino. Please keep in touch with @cmaglie

jimsy3, 2017-07-23:
what the heck is all of this jibberish...
here is how you delete a library:
https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment

per1234, 2017-07-23:
>what the heck is all of this jibberish...

@jimsy3, this "jibberish" is talking about adding a way to remove installed libraries ***via the Library Manager GUI***. Of course you have always been able to delete library files off your computer outside the Arduino IDE. That's completely off topic.

bperrybap, 2018-01-10:
The name is "Library Installer" it is "Library Manager" and part of managing your libraries includes removing them. Just saying.....

The board manager has an option to remove installed cores.
Not providing a way to remove libraries is just plain silly.

I could understand not providing a way to remove IDE bundled libraries as that could really screw things up but there should be a way to remove any library that is in the users sketchbook area.

I believe that people are over thinking this particularly if considering any complexities such as using some kind of trash bin or any kind of attempt to check on editing the library etc...
Just provide a remove simple button like what is in the boards manager and if there is a concern about screwing up accidentally removing an edited library just put in the typical warning dialog to get the user confirmation that they really want to remove.

For the time being a simple remove button for any sketchbook library with a conformation dialog should be good enough.

Longer term the library manager needs an entire revamp as the current methodology of using the users sketchbook for any/all library updates including the IDE bundled library updates is wrong and broken on so many levels.

per1234, 2020-03-16:
There was some additional good discussion on this topic in https://github.com/arduino/Arduino/issues/8269

rtek1000, 2020-03-16:
Well, It is very strange to be able to install and not to uninstall by the manager, instead of Manager we can call it "Installer and Updater". (#9879)

bperrybap, 2020-03-16:
I still believe that not having a way to remove/delete a library using the IDE is just plain silly, particularly for a tool called "Library Manager". 
I just don't understand the fear of adding this capability. 
How is forcing the user to go outside the IDE and use the operating system provided tools to remove the library directory better? How does it solve the concern:
 "Libraries managed by the Library Manager are mixed with those you've manually installed, maybe libraries you've written: making a mistake and deleting the wrong library is too easy."

IMO, that concern is just plain goofy.
While you can offer some protections, you can't always protect people from themselves.
Most users are not developing their own libraries. And these days, I'd be willing to bet that most less technical users are using the library manager to install them. So much of the concern seems overblown.

Perhaps there is a concern that the IDE seems to now be using the users sketchbook for updating bundled libraries. THAT is different issue based on making a terrible decision that needs to be solved independently of this issue.

It doesn't need to be complicated. Just put up a dialog with appropriate information and an warning.
If the user confirms, delete the directory. And only provide this capability for libraries in the sketchbook directory.
The dialog could display some information about library including:
- type of library (pre 1.x vs post 1.x)
- When the library was installed (based on the dates of the files)
- Library information from the libraries.properties file (if it exists)

And if there is still a concern about it, add a option in the Preferences/Settings that enables library management that enables this capability that must be checked to turn this capability on.

For example:
If you check the option to enable removing libraries  in settings, a [delete] button shows up on the existing library management page, when it detects a library installed/exists.
When the user clicks on [delete] the dialog with the information and the warning shows up.

At that point, to screw up, the user would have had to
-  enable the library management function from the settings/preferences
- selected an undesired library to remove/delete (which is only from libraries in the sketchbook)
- ignored/overridden a warning

That is 3 levels of protection.
Why can't this be enough?

rtek1000, 2020-03-16:
It is not necessary to add anything other than a file to serve as a flag.

If the user enters a library manually, this file will not be present in the library folder.

Then, the manager just needs to download the library (the same one that the user has access to) but create a file inside the library folder.

It is easier for the user to delete the wrong folder by opening the folder manager than the library manager.

bperrybap, 2020-03-16:
@rtek1000  It is more complicated than that to try to solve/prevent the potential issue(s) mentioned.

A user could have made changes to a library that was or could be installed using the library manager. Then there are users that are still doing manual installs using zip files. Often it is those manually installed libraries that need to be removed since it is easy to do and less sophisticated users have no clue where they ended up being installed on their filesystem.

IMO, any library that has been installed using the IDE, should be able to be removed/deleted using the IDE. 
Currently, that is not the case. And to me THAT is the main issue.

I still say it isn't worth trying to do any automated checks to try to prevent people from removing certain libraries for these corner cases.
I also believe that preventing the combinations of all the issues mentioned is simply not possible.
Just put a warning and remove the library if the user confirms.

Consider this, with the current  library manager. A user can install a library, make manual changes to it, then do an update and clobber/lose his changes when the updated library is installed.
And I've also seen cases where the IDE updates a bundled library by placing it the users sketchbook.
This is a terrible thing, as the user sketchbook is the users area and is not an area for the IDE use for its bundled library.  Doing this creates other issues particularly when there are multiple versions of the IDE installed.

So just to emphasize, the problem of clobbering a users library and the user losing his modified libraries already exists today.

rtek1000, 2020-03-16:
Well, if the whole problem revolves around the case that the user has made modifications manually. It should be possible to carry out a checksum of the original library, and if the value saved when installing is not the same when removing, then the user is prompted to make a backup, or not to remove it.

As stated, the IDE itself should be able to remove the libraries.

I don't know if everyone agrees, but I think that deleting is practically the same thing as updating, either way, it is no longer the same code that was stored after the update.

bperrybap, 2020-03-16:
> I don't know if everyone agrees, but I think that deleting is practically the same thing as updating, either way, it is no longer the same code that was stored after the update.

I agree, but I also think that the user should always be able to remove/delete, or update any library in the user sketchbook area using the IDE regardless of how the library got there or if it has been modified. To me that is what a tool called a "library manager" should allow the user do. (install, update, uninstall/delete libraries).

And this is why I think that all the concern about trying to prevent a user from removing or updating  a library in certain situations seems pointless. Not only is it subject to issues/errors that make mistakes that could prevent a user from being able to remove or update a library, but there can be legitimate reasons to want to remove or update a library even if has been installed manually or even manually modified since it was installed. (consider testing/experimentation, you install a library, do some tweaks or fix some bugs, then would like to simply re-install the stock version or update to a new version.)
Today you can overwrite a manually modified/edited library using the IDE library manager.
Today you can install a library, tweak it, fix a bug, and then when an update comes out to fix the bug, update to the newer library that will stomp on your modified library. 
(I have done this many times)

If the IDE library manager were to get "smart" and not allow a library to removed/updated if it were modified from its original form, then the user can no longer always manage the library from the IDE.

I believe in keeping things simple. 
Toss up a warning, and always let the user be the one to make the decision on what to do.

drmpf, 2020-06-13:
I have a related problem.
I installed a library via the manager  -- library shows as installed
I removed the library from the libraries dir -- restarted ide
manage still shows library as installed and won't let me re-install it.
How do I clear it from the library manager if I delete it manually

per1234, 2020-06-13:
@drmpf I'd be happy to help you, but I think the Arduino Forum would be a more appropriate place for that discussion. Please post your question over there. I'll be sure to see it and respond.

drmpf, 2020-06-15:
Actually I think there is a 'bug' in the library manager
What happens is this.
Install a local copy of a library (via zip)
Edit local copy and increase the version no
open Library Manager  and install library from there (previous version)
No error messages or any other messages.
Delete local library copy
Library Manager still says library is installed no way to either uninstall or re-install
After some searching found a directory
arduino_9802  containing the Library Manager installed version
deleting this directory removes the "installed" setting in the Library Manager and allows re-install.
SO What I think is needed is 
at least some message that the install did not complete
I assume it is failing on the final rename of the library dir from arduino_9802 to the correct name

Synetech, 2020-07-22:
> I've written a short explanation: _Since the IDE has no knowledge of your trash bin, we didn't implement a "delete" button._ Does it make sense?

Quite the contrary. If the IDE can't know what's okay, then why would the user be any better? The user is even _less_ likely to know what's safe to manually do or delete. 🤦

drmpf, 2020-07-22:
The IDE already scans libraries when doing a compile so it can check all the available libraries, either manually installed or installed via the library manager and update the library manager list accordingly

fscii, 2021-04-08:
Any progress on an "uninstall" option in the IDE Library Manager?
Its frustrating that I can add a library via button click but cannot uninstall one the same way.

Its like turning the key to start your car but having to open the hood to turn it off.

bperrybap, 2021-04-08:
@fscii 
Great analogy.

> Its like turning the key to start your car but having to open the hood to turn it off.

Yep, it seems silly to me that you can't use the IDE to remove a library was installed/added using the IDE.

I have seen cases of less technical users that are installing zip files from github. And given the way github constructs it zip images and the way the IDE zip install works, they can end up with multiple copies of the same library in different directories when trying to do updates.
I also see users install multiple copies of the same library or similar libraries. (LiquidCrystal_I2C is one example)
Both of these scenarios create additional issues from having colliding libraries.
The user created the mess using the IDE, and yet the user cannot use the IDE to remove libraries to clean it up.

I don't understand why there needs to be anything complicated implemented.
Why should the IDE add lots of complexity in an attempt to control what the user is allowed to do?
They are many scenarios where it is perfectly ok, and actually desirable to remove a modified library. 
Why does the IDE need to have such "fancy" capabilities around library removal?
Especially when currently there is still no way to remove a library at all.
Remember the KISS rule.

Fancy stuff adds complexity along with possibility of issues and, IMO, the added capabilities are of questionable value, particularly since I believe it will never be possible to catch all the corner cases.
And even if all the corner cases could be resolved, in the end, users will need the ability for an override to allow users to get rid of a library they are done with or perhaps just to get rid of a library so they can start over clean.
i.e. the ability to tell the library manager:

>  YES I know what I'm doing, I know this library has been modified. I don't care. Please delete this library anyway

And once you have this override control capability, aren't you right back to having something much simpler?
So why bother having done all the fancy stuff?

The primary need is the ability to remove libraries using the library manager and that capability does not exist today.

Put up a warning dialog asking for conformation and state that uninstalling / removing a library cannot be undone and once removed, it is gone.
i.e. Warn the user but yet let the user be in charge of what happens.
Why isn't that an acceptable solution?

It is very simple to implement and yet provides a needed library management capability.

In the mean time, it has now been over 6 years of having a library manager and yet there is still no way to remove / uninstall a library using the IDE library manager.

rtek1000, 2021-04-08:
Regarding the conflict between libraries, it might be interesting to have a way to select the library in use.

B4X compilers (B4J, B4A, B4R ...) have a list of available libraries, which the IDE identifies.
[image]

Mikroelectronica compilers (MikroC, MikroBasic ...) also have this type of library list.
[image]

per1234, 2024-04-30:
The capability for uninstalling libraries via Library Manager has been implemented in Arduino IDE 2.x:

1. Select **Sketch > Include Library > Manage Libraries...** from the Arduino IDE menus to open the "**Library Manager**" view in the left side panel.
1. Find the entry for the library you wish to uninstall.
1. Hover the mouse pointer over the entry.
1. You will see a **●●●** icon appear near the top right corner of the library entry. Click on that icon.
   A context menu will open.
1. Select "**Remove**" from the context menu.
   An "**Uninstall**" dialog will open.
1. Click the "**YES**" button in the "**Uninstall**" dialog to confirm that you want to uninstall the library.
   The dialog will close.
1. Wait for the uninstall process to finish, as indicated by a notification at the bottom right corner of the Arduino IDE window:
   > **ⓘ Successfully uninstalled library ...**

closed 2024-04-30 by per1234

bperrybap, 2024-05-01:
I tried this on 2.3.2 and it kind of works.
When I search for a library, and it finds it.
The library shows up but there is a [REMOVE] button next to the box that shows the currently installed version.
I don't have to use a context menu.

However, the library manager does not show all the libraries that are in my sketchbook/libraries directory.
For example there are some cases where I can search for library and it doesn't find it.
It also doesn't see some libraries when you just scroll through the list without doing a search.
However, I can use sketch->include and it finds it.
So the library manager must be trying to do something too smart for it's own good.
Maybe it only allows searching for libraries it installed? 
Not sure what is happening but the library manager should allow you to remove any library in sketchbook/libraries regardless of how it got there.
I'll have to play with the library manager some more to figure out why it is not working for some libraries.
 Seems like a bug. 

Also, if you do something like have a copy of a library in sketchbook/libraries
(copy and paste the library directory to create a "copy" directory of the library, example copy past the foo library to create the "foo (copy)" directory) the library manager won't show this library either so you can't use the library manager to remove a copied directory in the sketchbook/libraries directory.
Seems like another bug.

Seems like the GUI could have been a bit different to make things simpler.
i.e. separate installation from removal.
For removal, it could show ALL the directories/libraries under sketchbook/libraries and then allow you to remove anyone you want. 

I'll play with it some more but it does seem like in the existing GUI there are some library manager bugs/issues and they affect more than just removal.

per1234, 2024-05-01:
>The library shows up but there is a [REMOVE] button next to the box that shows the currently installed version.
>I don't have to use a context menu.

[TMTOWTDI](https://en.wiktionary.org/wiki/TMTOWTDI). The "**REMOVE**" button will only be there if you have the latest version of the library installed. If you have an older version installed then the button will instead be "**UPDATE**". I chose the universal and most user friendly (IMO) procedure for my instructions but you are welcome to use whatever procedure you prefer.

> Maybe it only allows searching for libraries it installed?

Yep. That is the intentional design.

> Also, if you do something like have a copy of a library in sketchbook/libraries
(copy and paste the library directory to create a "copy" directory of the library, example copy past the foo library to create the "foo (copy)" directory) the library manager won't show this library either so you can't use the library manager to remove a copied directory in the sketchbook/libraries directory.

If you manually installed the library then you can very well manually uninstall it as well.