arduino/Arduino #2880: Difference between "contributed" and "recommended" libraries is unclear.

damellis, 2015-04-02:
I'm confused as to the difference between the "contributed" and the "recommended" categories in the library manager. From the names, I would guess that "recommended" libraries are more strongly endorsed than the ones that are simply "contributed". But based on the fact that the "recommended" category comes last in the drop-down menu in the dialog and that the "recommended" list contains many more libraries than the "contributed" list, I would guess the opposite (i.e. that the "contributed" libraries are more selective and therefore more highly endorsed than the "recommended" ones). It would be good to try to make this distinction more clear, or maybe remove it altogether?

ffissore, 2015-04-02:
I'll change the order and make Recommended appear before Contributed

ffissore, 2015-04-02:
BTW, yes the difference between the two is the level of official endorsement

rogerclarkmelbourne, 2015-04-04:
I'm not sure how may normal Arduino users will be able to determine which name represents what level of endorsement.

Its great that people can easily get loads of libraries, but the downside, is do all libraries listed work will all listed hardware.

I suspect this is a high livelihood that a lot of libraries don't compile or work with the Due.

I can foresee a lot of postings to the forum asking why xxx library that appears to be part of the IDE doesn't seem to work with yyy board.

I see some libs have "Due Only" in their description, which is OK, but really, there needs to be some sort of dependency file attached to the library which specifies which hardware it works with

PaulStoffregen, 2015-04-04:
This scheme, with a casual process for "recommended", leads to some unexpected results.

For example, it seems all Adafruit libraries are recommended, even if their purpose is very narrow.  One example that I find really amusing is MAX31850_OneWire is "recommended", even though it's based on an older copy of OneWire with Adafruit's only apparent addition being a couple more examples specifically for that chip.  Meanwhile, the newer version of OneWire isn't recommended!

rogerclarkmelbourne, 2015-04-04:
With so many libraries and now ones being submitted all the time, its also had to see the wood for the trees. 

Also, there are also what appear to be duplicate libs e.g. ILI9341

ffissore, 2015-06-04:
As you have seen during these 2 months, we don't control/moderate/other the library devs ask us to include. If they conform to the [spec](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) (and the [limitations](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#limitations) of the job), they are included.
This gives room to duplicates, for better or for worse.
As for the difference between recommended, contributed, arduino (and soon "at heart"), it reflects the official endorsement Arduino has with the authors

closed 2015-06-08 by ffissore