arduino/Arduino #2881: Add Boards via Zip file

NicoHood, 2015-04-02:
The boards manager also needs a button to add .zip board files as the library menu.

rogerclarkmelbourne, 2015-04-02:
A specification will need to be written and discussed for this.

E.g some boards are AVR and only need their cores files

Some will be ARM and can use the ARM compiler that may already be installed for the Due, e.g. STM32 

Some will need both tools (compiler) and cores e.g esp8266 (xtensia - MIPS compiler)

Otherwise either the core compiler perhaps would get overwritten by a incompatible version e.g. For ARM, that had different settings which were not compatible with the Due

(I'm currently trying to fix issues in the xtensia gcc for esp8266, so that libiconv-2.dll is statically linked, because if it is not statically linked, I have to copy libiconv-2.dll to the installation root of the IDE :-(
Hence I can't make a file set that would be installable

So I can foresee issues if someone compiles and packages their own ARM compiler which has settings which are incompatible with due, but installs it before installing the due

I guess these occurrences would be rare, but are probably worth considering at the start of this feature.

NicoHood, 2015-04-02:
I was thinking of just unzipping the zip into sketchbook/hardware/
Using AVR only though.

rogerclarkmelbourne, 2015-04-04:
Unzipping there is fine for AVR as the AVR compiler is installed by default in its original location.

It only effects ARM based boards as the ARM compiler is not longer installed by default, and when the Due is installed, the ARM compiler is in a totally different (and IMHO a wrong location) i.e binary files in AppData which is the folder intended for user preferences.

ffissore, 2015-05-18:
Since the IDE supports multiple package_index urls (some of them [listed here](https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls)) and since installing hardware into sketchbook "hardware" folder is a well known and documented feature, may we close this issue?

NicoHood, 2015-05-18:
You have a button to add libraries via Zip file. Why not add the same function for boards? That would make it much simpler for people to install the cores which are not in the boards manager. People are always confused how to install it, where to move it and if its the first time there is nor hardware folder at all, only libraries. So this would be a nice addition. Shouldnt be too hard to implement I guess. Just the same as the libraries, in the hardware folder instead.

ffissore, 2015-05-18:
AFAIK there are no cores available as zip files. Only zip files with some instructions. A core developer should rather go for a structured way (json file, SHA sums and so forth) in order to provide a seamless user experience, rather than hoping to have zipped the file correctly and leaving it up to users.
With Boards Manager and its ability to fetch from github, I can't see a valid reason for a core dev to pack a zip file rather than being listed in boards manager

NicoHood, 2015-05-20:
Then you could also remove the zip feature for the libraries X_x

ffissore, 2015-05-20:
I would like to, but it's already there and it's working

NicoHood, 2015-05-20:
I think you can mark this issue as "wontfix" or even close it.

closed 2015-05-20 by ffissore

aboshaddad, 2020-05-30:
I did something should be work to install board ZIP file offline and manually:
[https://deve-sp.blogspot.com/2020/05/zip-boards-manager-how-to-install.html](https://deve-sp.blogspot.com/2020/05/zip-boards-manager-how-to-install.html)

TToeyIT, 2021-04-28:
> I did something should be work to install board ZIP file offline and manually:
> https://deve-sp.blogspot.com/2020/05/zip-boards-manager-how-to-install.html

Thank you, It's work for me! I've been looking for this solution for a long time.