arduino/Arduino #2883: There's no way to only close a tab (and nothing else)

q2dg, 2015-04-02:
I've already said it in #2350, but as it's not exactly the same issue, I open a new one:

*There's no option in rightest drop-down menu responsible for closing a tab without deleting its content from disk (in other words: there is a "delete" option but not a "close" option)

*File-> Close and File->Quit do the same: they close the IDE...(I think former should close only active tab)

Tested on 1.6.1 and on 1.6.3

ffissore, 2015-05-29:
How are you supposed to reopen the closed tab?

q2dg, 2015-05-29:
Good question. Maybe an "Open file as new tab in current project" option is missing, too.

ffissore, 2015-05-29:
This would complicate the UI. We'd need a file tree on a side panel, but we won't have one. See #1590

closed 2015-05-29 by ffissore

q2dg, 2015-05-29:
Well, I don't think so.Simply opening the FileChooser dialog should be enough.

But, anyway, my original concern was that closing a tab deletes the file. I think this behaviour is very risky and confusing

ffissore, 2015-05-29:
I disagree. Label says "delete", not "close". Deleting is always risky, but the label should not be confusing

stevenkaza, 2016-04-21:
Please revisit this, it makes working with the IDE a pain!

borgdrone7, 2016-10-30:
This "feature" makes Arduino IDE completely unusable, will try to find other way to develop.

Kadah, 2017-03-01:
The tab behavior is confusing and counter to the general notion of how tabs in virtually every other IDE or source/text editor.
Example:
If  you manually open a file into a tab by drag and drop to view it, like a library source file. What I would expect having used any other IDE or source editor with tabs is that the file is being opened, but what actually happens is that the file has been copied in the file into the project folder and that copy has been opened in the IDE via the "every file in the project folder is a tab" feature.

devyte, 2017-03-01:
I also would like to revisit this. Any half way serious project splits the code over various files. In the Arduino IDE, this means one tab open for each .h and each .cpp file. Having 20+ tabs open makes the IDE very unusable.
I don't think a tree view is necessary, like it was stated some time ago. Just allowing a tab to close without deleting should be sufficient.
If the issue is how to know which files belong to the project, without actually having a project file that lists them, I suggest for now just building all files that are in the same dir as the .ino. You could potentially also look in subdirs, but that could be done later.
I think this would improve the usability tremendously for minimal effort.

Kadah, 2017-03-06:
To be fair, up till recently, a sketch with 20+ source files was unlikely to fit on an ATMega outside of the 2560. Now with the support for a number of 32bit MCUs, very larger sketches are entirely possibly.

Airheadbit, 2017-09-07:
The whole 'Close' issue is not user friendly, at this point I'm looking for something else that 1: Has standard Open and Close features 2: Has documentation that matches the product. SerialUSB is an example.
If you're going to protect the dummies by denying them the  ability to close a file they will not learn and when too many files fill all the tabs because you can't close tabs or files the the application is useless.

optio50, 2017-11-23:
Whats wrong with adding a close tab option? Seems kinda messed up.

mfoley40, 2017-12-18:
I'd also like to see a close tab available. Then, a corresponding open tab option which allows the selection of a file in the directory with the .ino file. Building and everything else would remain the same.

PalantirWielder, 2018-01-23:
NOTE for others viewing this in the future:

When opening a sketch, the IDE _automatically opens all relevant('.c', '.h', etc) files that reside in the sketch (root) folder_. Therefore to "remove" a tab/file, file(s) need to be someplace other than that folder. For example if you create a sub-folder and move files there, then they will not show up when you reopen the Sketch.

This can be done temporarily as long as you don't need to compile those files.

bkelly13, 2019-09-23:
I wrote another issue on this, then found this one.  Adding my support:
The fundamental idea that tabs cannot be closed is just wrong.  Wrong to the point of being outright absurd.  Yes, I have used many IDEs, beginning with the very first one, Turbo Pascal in the early 1980s, through many issues of Visual Studio, and including some Linux versions.  
Please provide the ability to simply close a tab.  And DO NOT make the IDE or me delete that file.

PaulStoffregen, 2019-09-23:
Who is to say what is & is not "just wrong"?

bkelly13, 2019-09-23:
Who? …  Who Indeed ??? !!!

 

For one:  I am!

 

I and just about everyone one else that uses any widely used IDE.

And any and probably everyone that uses any tab’ed dialog.

 

Have you ever used any recent web browser?  Anything in the last, maybe ten years or so?

Open and close a tabs there.  Did anything get deleted?   The answer is no.

 

Go to gedit on Linux and open multiple files.  You get tabs.  Close a tab.  If the text was changed, it will ask if you want to save.  Then it closes the tab.  Be sure to note:  It does not delete the file.  It closes the tab.

 

Have you ever used Visual Studio.  It will open files, tab them, and close the tabs.  Try it some time.

 

Can you reference ANY application that allows the user to open tabs then deletes files on closure?

I think that answer is No.

Maybe you can find one, but it will be the exception.

 

Get real here!  Stop being defensive!  This is not an attack upon you.

The IDE has a fundamental flaw.  

Yes, we can work around it.  That does not change the situation that it has a fundamental flaw.

 

Bryan

Perlovka, 2019-10-05:
"Legitimate, unique issue reports are certainly welcome here. They are a valuable contribution to the Arduino project." © @per1234 

Bullshit.

PaulStoffregen, 2019-10-05:
Other IDEs like Visual Studio provide a file display to show which files are part of your project.  If you remove a file from that part of the GUI, even if the file isn't deleted, it is no longer compiled and linked into your project.

Arduino's very simple & minimalist doesn't provide any other way to configure or visualize which files are actually part of your project.  All the files located in the folder are always used.  Arduino's build system doesn't support source code files in the folder that aren't used.  If a file is present in the folder, it will be compiled and used.

So if Arduino were to implement simply closing of tabs without changing the underlying files, when those files are compiled and used, wouldn't that also be considered "just wrong"?

If there is any sort of an error in any of those files, what sort of behavior should the GUI offer?  Would the file be opened again (against the user's wishes), for the sake of highlighting the line where the first compile error occurred?  Or should the console get an error message with the filename, but the GUI wouldn't be showing that file anymore?

Perlovka, 2019-10-05:
Simply add _possibility_ to close the tab, and to save it, if it was changed. And yes, "should the console get an error message with the filename" will be good enough. Compiler does this by default, isn't it?

Keep it simple. Every time a simple feature request becomes "very complex task" and being closed as wotfix, even when the whole thing is to change 15 strings of code.

bkelly13, 2019-10-08:
RE: Arduino's very simple & minimalist doesn't provide any other way to configure or visualize which files are actually part of your project. All the files located in the folder are always used.

 

In my not so humble opinion, and in this day and age, that is just too minimalist.  Nothing should be a part of the project until the user takes some level effort to make it so.  And any file should be eligible for exclusion with the same minimal effort.

ckohrt, 2021-01-06:
I just wrote a larger program and I cannot edit the files anymore, because they are out of sight --> the tabs are not visible anymore.

I'd like to make them invisible. I see the discussion is long, so it might be the time to change the IDE now because larger programs are not supposed to be done with this IDE?

GordonLElliott, 2021-02-09:
When is someone going to figure this out. Even if there is a way to close a tab without closing the entire Arduino app, I can't easily find it.

This should be obvious, easy to find. (And if, as some suggest, there is no way at all to close a tab--that is a failure that needs to be remedied.)