arduino/Arduino #2884: Library and Board managers haven't close button (X) in Fedora 21 (64 bits)/Gnome 3.14

q2dg, 2015-04-02:
...So I can't close its modal windows unless I right-click over it to show system contextual menu and select "Close" from it. It's a nuisance

ffissore, 2015-04-02:
This is new to me: can you attach a screenshot?

q2dg, 2015-04-02:
Yes!
Here they are
[image]
[image]

ffissore, 2015-04-02:
Ok. Wow. No buttons. Uhm.. Anyway, if you press ESC once or twice, the modal will close

ffissore, 2015-04-02:
It's a workaround. I'll see where I can put a close button

q2dg, 2015-04-02:
Ok. Thanks!

sej7278, 2015-04-03:
Same on Debian Jessie 64-Bit with Gnome 3.14, escape works, but there's no icon

ffissore, 2015-06-05:
Fixed. Fix will be available with the next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit c754481e (2015-06-05): Boards/Lib manager: added "Close" button at lower right corner, so users with ungrateful window managers can close that modal using the mouse. Fixes #2884
closed 2015-06-05 by ffissore