arduino/Arduino #2886: 1.6.3 - platform.txt avrdude Custom Config Issue

toddtreece, 2015-04-02:
I can't seem to change the `tools.avrdude.config.path` in the platform.txt file for custom boards. Is this something I should be able to override? It always points to the default config:

```
-C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf
```

agdl, 2015-04-03:
sorry wrong click on it

clefranc, 2015-04-05:
Sorry if this not the right place to ask, but my issue seems related. Can't change the optimization `compiler.c.flags` from `-Os` to `-O3` in platform.txt for AVR, ie when compiling still shoes -Os. Was correct with 1.6.2.

clefranc, 2015-04-07:
So, is this a bug or a new functionality?

cmaglie, 2015-07-03:
Hi @clefranc 

you can change any platform variable in your custom `platform.txt` or `boards.txt` (and override the "default" value in the referred core).

With the latest IDE the `compiler.path` variable is defined in platform.txt as:

```
compiler.path={runtime.tools.avr-gcc.path}/bin/
```

you should be able to change to anything else that works for you.

clefranc, 2015-07-03:
@cmaglie Hi Cristian, thank you for your interest in the resolution of this issue.

Like I said in my previous comment, when I change ```compiler.c.flags``` in ```platform.txt``` from ```-Os``` to ```-O3```, it is ignored when compiling. In fact, I'm setting ```compiler.c.elf.flags``` and ```compiler.cpp.flags``` to ```-O3``` as well. The result is still compiled with the ```-Os``` optimization option.

Prior to 1.6.3, the optimization option was compiling correctly to my ```-O3``` setting. Something was introduced in 1.6.3 that affect ```platform.txt``` and that I can't understand:

```
ARDUINO 1.6.3 - 2015.04.02

[ide]
* Fixed an ArchiveExtractor issue when dealing with native stuff, affecting some linuxes
* MacOSX: MACOSX_BUNDLED_JVM property is automatically set to JAVA_HOME if not provided
* New splashscreen image
* New editor theme
* MacOSX and Windows: https links were not working, fixed

[core]
* Upgraded toolchains: statically compiled, ship with Link Time Optimization plugin (disabled by default)
* Restored backward compatibility by providing a platform.txt in-memory rewrite: old values (eg: compiler.path) are converted to new values
* platform.txt rewriting can be disabled if target platform has a "rewriting=disabled" key/value pair set
* AVR bundle files (core files and toolchain) is again inside IDE "hardware" folder, instead of unpacked into user's home folder at first IDE startup
* Advise of duplicate libraries after compiling. Thanks @PaulStoffregen
```

All I want is output compiled code with modified optimization option, I can't find a way to do that with 1.6.3 and above.

matthijskooijman, 2020-10-08:
@toddtreece, @clefranc, I came across this old issue. A lot has changed in the arduino building landscape since it was reported, so I wonder if this is still a problem. Could you make try with a recent version and report back here? If this has been solved already, I'd like to close this issue (and if not, maybe figure out what the problem is exactly).

per1234, 2021-03-28:
Closing due to lack of response.

@toddtreece @clefranc if you'll provide the feedback Matthijs requested I'll be happy to re-open this issue.

closed 2021-03-28 by per1234