arduino/Arduino #2887: CapacitiveSensor library (https://github.com/arduino-libraries/CapacitiveSensor) isn't listed in Library Manager

q2dg, 2015-04-03:
I mean...if adding it to default list of installed libraries isn't worth, I think at least could be useful having the choice of installing it through Library Manager. 

Maybe Paul's one (https://github.com/PaulStoffregen/CapacitiveSensor) too/instead.

Thanks a lot

PaulStoffregen, 2015-04-03:
The copy of CapacitiveSensor I've been maintaining is already available through the new library manager.

q2dg, 2015-04-03:
Sorry, but in 1.6.3 I can't find it

PaulStoffregen, 2015-04-03:
I just looked again, and indeed it seems to have disappeared.

ffissore, 2015-04-03:
yeah sorry, I didn't add it because of a clash with the other CapacitiveSensor lib. @cmaglie is in charge of choosing one

PaulStoffregen, 2015-04-03:
Just posted this on the mail list, answering a similar question from a couple days ago:

*Regarding CapacitiveSensor, my repository should probably be used.  The copy on Arduino's repository has an older port to Arduino Due, with only some of the macros converted to native I/O.  My copy uses fully native I/O on Due, and it also supports more boards.*

cmaglie, 2015-04-08:
@ffissore 
let's use Paul's fork.

cmaglie, 2015-04-08:
and please, do not remove arduino-libraries/CapaciveSensor because its actually used in our tutorials and contains two libraries:
- an old copy of `CapacitiveSensor
- a `CapacitiveSensorDue` (made by @Liarco).

Probably the best way to solve this is:
1) remove `CapacitiveSensor` from arduino-libraries/CapacitiveSensor (and keep only `CapacitiveSensorDue`)
2) remane  `libraries/CapacitiveSensor` to `libraries/CapacitiveSensorDue`
3) fork PaulStoffregen/CapacitiveSensor to `arduino-libraries/CapaciveSensor` to keep our tutorials updated.

ffissore, 2015-04-08:
@PaulStoffregen CapacitiveSensor is now listed. Thanks!

closed 2015-04-08 by ffissore

PaulStoffregen, 2015-04-08:
My copy supports Arduino Due.

q2dg, 2015-04-08:
If so...I think there's no need to maintain arduino-libraries repo, right? I think it's more suitable rewriting some lines of some tutorials than having to dance between two similar-but-not-so libraries.

cmaglie, 2015-04-08:
> My copy supports Arduino Due.

I know, I've already choose your (and **only** your) library to be included in the library manager.

My concern is that in the arduino-library/CapacitiveSensor there is a CapacitiveSensorDue lib made by @Liarco:

https://github.com/arduino-libraries/CapacitiveSensor/tree/master/libraries

now, whatever library is better, my concern is that this is the only copy available in a public repository. Before deleting the only existent copy, my proposal to rename it until we understand what @Liarco wants to do with his library.

@Liarco may you comment on this matter?

Liarco, 2015-04-08:
Thank you for taking the CapacitiveSensorDue library into consideration.
It has been written many time ago and I was a complete newbie in the Arduino world, I bet Paul's fork should be far better than mine, I would opt for that to be the only library on the public repository.

cmaglie, 2015-04-10:
Thanks @Liarco for answering!
I've removed the old CapacitiveSensor and replaced it with a fork from @PaulStoffregen version.

PaulStoffregen, 2015-04-10:
FYI - I've been holding off updating CapacitiveSensor and a few other libraries, until Arduino Zero is properly released.  The last thing I want to do is help out those "other" guys!

When Zero is released by Arduino.cc, I'll update CapacitiveSensor and OneWire, which I also maintain.