arduino/Arduino #2890: Arduino 1.6.3 on Mac can't find EEPROM library

sej7278, 2015-04-03:
Linux version works fine, and on the Mac it does seem to be installed in:

/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM

But the IDE isn't finding it.

If I use [arduino-mk](https://github.com/sudar/Arduino-Makefile) it builds fine.

ffissore, 2015-04-03:
Can you paste compilation verbose output ?

sej7278, 2015-04-03:
It must be a problem with the arduino-tiny core as it works with an uno as well as @damellis attiny core.

```
Arduino: 1.6.3 (Mac OS X), Board: "ATtiny85 @ 1 MHz  (internal oscillator; BOD disabled)"

Third-party platform.txt does not define compiler.path. Please report this to the third-party hardware maintainer.
Build options changed, rebuilding all

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny85 -DF_CPU=1000000L -DARDUINO=10603 -DARDUINO_AVR_ATTINY85AT1 -DARDUINO_ARCH_AVR -I/Users/simon/sketchbook/hardware/tiny/avr/cores/tiny /var/folders/ws/gysqz2dx1lzbd5x9mvxl3yfm0000gn/T/build3060276760662295255.tmp/gears.cpp -o /var/folders/ws/gysqz2dx1lzbd5x9mvxl3yfm0000gn/T/build3060276760662295255.tmp/gears.cpp.o 
gears.ino:1:20: fatal error: EEPROM.h: No such file or directory
compilation terminated.
Error compiling.
```

p.s. why have we changed the arduino version string, it used to just remove the dots, now its zero-padding too? shouldn't it be ```-DARDUINO=163``` ?

ffissore, 2015-04-03:
EEPROM is one of those few libs included in the avr core, so yes it's an issue with arduino-tiny.
As for version, it's padded to allow version numbers like 1.6.10, which weren't just usable with the old syntax. See this discussion on dev list https://groups.google.com/a/arduino.cc/d/msg/developers/M49P_1GIeVo/chi2ak51CTIJ

closed 2015-04-03 by ffissore