arduino/Arduino #2892: 1.6.3 fails to list serial ports in ubuntu 12.04

bgodard, 2015-04-03:
IDE 1.6.3 starts but doesn't show ports. 
Tools->Port menu greyed out
Messages in console when IDE starts:

cybernetics@clarke:~/arduino-1.6.3$ ./arduino 
Error parsing gtk-icon-sizes string: ''
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.NullPointerException
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:143)
	at processing.app.linux.Platform.resolveDeviceAttachedTo(Platform.java:141)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:73)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
cybernetics@clarke:~/arduino-1.6.3$ 

ADDITIONAL INFO:

cybernetics@clarke:~/arduino-1.6.3$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.2 LTS
Release:	14.04
Codename:	trusty
cybernetics@clarke:~/arduino-1.6.3$ uname -a
Linux clarke 3.13.0-39-generic #66-Ubuntu SMP Tue Oct 28 13:31:23 UTC 2014 i686 i686 i686 GNU/Linux
cybernetics@clarke:~/arduino-1.6.3$ java -version
java version "1.7.0_75"
OpenJDK Runtime Environment (IcedTea 2.5.4) (7u75-2.5.4-1~trusty1)
OpenJDK Client VM (build 24.75-b04, mixed mode, sharing)
cybernetics@clarke:~/arduino-1.6.3$ lsusb 
Bus 001 Device 013: ID 050d:705a Belkin Components F5D7050 Wireless G Adapter v3000 [Ralink RT2571W]
Bus 001 Device 019: ID 2341:003d Arduino SA 
Bus 001 Device 020: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 0a5c:201e Broadcom Corp. IBM Integrated Bluetooth IV
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

wayoda, 2015-04-03:
Is the device file for the board created when you plug it in?
The Unix cmd `dmesg`should print something similar to this after the due is plugged in
```
wayoda@shredder:~$ demsg

[28042.136652] usb 3-2.5: new full-speed USB device number 20 using xhci_hcd
[28042.243179] usb 3-2.5: New USB device found, idVendor=2341, idProduct=003d
[28042.243184] usb 3-2.5: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[28042.243187] usb 3-2.5: Product: Arduino Due Prog. Port
[28042.243189] usb 3-2.5: Manufacturer: Arduino (www.arduino.cc)
[28042.243191] usb 3-2.5: SerialNumber: 74134373733351D00282
[28042.243398] usb 3-2.5: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[28042.243750] cdc_acm 3-2.5:1.0: ttyACM0: USB ACM device
```

bgodard, 2015-04-04:
Yes the device /dev/ttyACM0 is correctly created. Also the port listing should work whether or not an Arduino is connected.

cybernetics@clarke:~/arduino-1.6.3$ ls -all /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 Apr  4 15:03 /dev/ttyACM0
cybernetics@clarke:~/arduino-1.6.3$ groups
cybernetics adm dialout cdrom sudo dip plugdev lpadmin sambashare

Now it works sometimes when I start the Arduino IDE and sometimes not. It seems to be a bit random except that as of now after a boot once it starts correctly it continues starting correctly until next reboot but the first Arduino IDE start(s) after a reboot fails to list the ports (to take with a grain of salt: I have not done enough boots/Arduino IDE start to generalize).

Now I have other problems:
-  serial port configuration requiring very frequent manual set of the baud rate (shouldn't the board programmer do that?) before loading the program as described in http://forum.arduino.cc/index.php?topic=148613.0. This is also a bit random and might be related to the problem above.
- no file MouseController.h (Due usb host lib) in 1.6.3 bundle or in extension installed by boards manager for SAM boards (Due). Where do I find this?

hemantsangwan, 2015-04-08:
Getting the same issue in Ubuntu 14.

PaulStoffregen, 2015-04-08:
Is the failure only temporary?  In other words, if you go get a cup of coffee, when you come back and try again, does the Tools > Ports menu later have your device?

bgodard, 2015-04-08:
When this exception occurs at Arduino IDE startup, Tools->Ports menu is
greyed out. I have only been able to solve this by restarting the IDE until
it works (in that sense it is temporary).
I have had this problem with 2 PCs under 14.04 with IDE 1.6.3. It does NOT
seem to be related to which Arduino is connected (Due, Uno or None) but on
one of the PC, I have seen it happen only once out of about 10 IDE startup
while on the other it was more like 50/50.

niallp, 2015-04-08:
Seeing something similar here ... on ubuntu 12.04, sometimes get the SerialBoardsLister error on startup in which case no ports available. I've been try various IDE versions, the old one from the repos always works (but doesn't support the Yun) while 1.5.8 and 1.6.3 work intermittently. It seems like once one works, the other won't, maybe file locking ?

natanael-fim, 2015-04-12:
when happened this with me i do:
$ sudo usermod -a -G tty MY_USERNAME
$ sudo usermod -a -G dialout MY_USERNAME

- logout;

- logon; 

DONE :)

niallp, 2015-04-13:
Unfortunately not enough for me, I was already in the appropriate 
groups. I haven't found the magic conditions for 1.6.3 to work 
consistently on the serial port, also turns out to not see a device on 
the LAN either, only 1.5.8 works there so I'll have to stick with the 
older version for now.

closed 2015-04-13 by ffissore

ffissore, 2015-04-13:
Probably this issue has the same root cause of #2837. It should be fixed in the latest nightly (just triggered). Please download and try it http://arduino.cc/en/Main/Software#nightly

niallp, 2015-04-13:
Thanks, the nightly build (marked 1.6.4) works for both serial discovery and network.

misteralex, 2015-04-14:
Serial monitor in 1.6.4 works fine on Linux 32 bits over /dev/ttyACM0 (virtual Debian context).

phenom128, 2015-05-13:
delete the serial.* lines in ~/.arduino15/preferences.txt
serial.parity=N
serial.port=/dev/ttyUSB0
serial.stopbits=1

no exception from the serial detect starting from a terminal window

phenom128, 2015-05-13:
apols debian 8 & arduino 1.6.4
i edited the preferences.txt deleting the serial.* lines and on restarting the ide
it generated new lines for the serial.
i did the same when i had greyed out port, thati had a preferences.txt with a windows style com port reference

Vinayakj009, 2017-05-03:
Hi. I saw something peculear happening in arduino-1.8.2, on ubuntu 16.04.02 LTS.
If I run ./arduino from the arduino home directory, while it gives the exception 
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
	at java.util.LinkedList.get(LinkedList.java:476)
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:241)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.retriggerDiscovery(SerialBoardsLister.java:113)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:183)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

The peculear part is, it gives this exception only once and when the usb device is connected.

Also if I run the command without the device connected, this error does not show up untill I attach the usb device.

I have had this problem before, and sometimes it does not happen. Also when it does, my pc becomes useless for arduino for days togather.

Note:
I tried all the tricks.
Adding user to dialout group,
running arduino as sudo
connecting disconnecting the usb port several times
etc etc

TCTKeerthiga, 2017-07-06:
My application provide below exception:
 java.lang.RuntimeException: java.io.IOException: jssc.SerialPortException: Port name - COM1; Method name - openPort(); Exception type - Port not found. 

Anyone know how to fix this exception ?

per1234, 2017-07-06:
@TCTKeerthiga generally this problem is caused by forgetting to select the port of your Arduino from the **Tools > Port** menu.

Your problem is more appropriate for the Arduino Forum so let's continue any further discussion there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.
3. Add a comment to this issue with the link to your thread to help anyone else who later has the same problem find the solution.

TCTKeerthiga, 2017-07-07:
@per1234 : Thanks for your comments,Its very useful,
I have selected the currently running port in my application then run successfully.

commit 350adb99 (2015-04-13): Also SerialDiscovery was affected by bug found at 40535dfc5560e2bbdb2c9479d7b6b43ad3ae4234. Fixes #2892